neurodroid/stimfit #10: Stfio module build fails on linux

Taomach, 2015-01-12:
Hello!
I am trying to build the stfio module using the build script that used to work before. The build fails with the following error:

    /usr/bin/swig -c++ -python -o ../../../../src/pystfio/pystfio_wrap.cxx ../../../../src/pystfio/pystfio.i
      CXX      libpystfio_la-pystfio.lo
    ../../../../src/pystfio/pystfio.i:389: Error: Unknown SWIG preprocessor directive: code
    ../../../../src/pystfio/pystfio.i:437: Error: Unknown SWIG preprocessor directive: Guess

I use Arch Linux x86_64, python 3.4.2, hdf5 1.8.14, numpy 1.9.1, swig 3.0.3, boost 1.57.0. The build script can be found here: https://aur.archlinux.org/packages/py/python-stfio-git/PKGBUILD

neurodroid, 2015-01-12:
Should be fixed by the commit references above - seems like your version of swig gets confused by comments within Python code in interface files and tries to parse them as preprocessor directives. Let me know if this fixes the issue so that I can close it.

Taomach, 2015-01-19:
Yes, this change fixes the build. Thank you.

closed 2015-01-19 by neurodroid