neurodroid/stimfit #11: std::vector index out of range reading gap-free abf. It seems that m_pFI->m_uLastEpiSize may never be set.

kh90909, 2015-01-21:
I've run into another of these `std::vector::operator[]` index out of range checks that seem to be enabled only in VS2008 builds. However, I think this one might point to an underlying bug.

When reading a gap-free file larger than `maxsize`, [`ABF2_GetNumSamples()`](https://github.com/neurodroid/stimfit/blob/b8326dbafa01ace1eb115ae84c1f9a236bd0fac1/src/libstfio/abf/abflib.cpp#L214) appears to always set `uNumSamples` to zero for the last episode. This results in a zero length `TempSection` vector being passed to [`ABF2_ReadChannel()`](https://github.com/neurodroid/stimfit/blob/b8326dbafa01ace1eb115ae84c1f9a236bd0fac1/src/libstfio/abf/abflib.cpp#L230), which triggers the index out of range error when passing `&pfBuffer[0]` to [`ConvertADCToFloats()`](https://github.com/neurodroid/stimfit/blob/b8326dbafa01ace1eb115ae84c1f9a236bd0fac1/src/libstfio/abf/axon/AxAbfFio32/abffiles.cpp#L2348).

For the last episode only, [`ABF2_GetNumSamples()`](https://github.com/neurodroid/stimfit/blob/b8326dbafa01ace1eb115ae84c1f9a236bd0fac1/src/libstfio/abf/abflib.cpp#L214) calls [`pFI->GetLastEpiSize()`](https://github.com/neurodroid/stimfit/blob/b8326dbafa01ace1eb115ae84c1f9a236bd0fac1/src/libstfio/abf/axon/AxAbfFio32/abffiles.cpp#L3344) to get the number of samples, which in turn returns `m_uLastEpiSize`. As far as I can tell, `m_uLastEpiSize` is not getting set anywhere, but I'm not familiar enough with the code to know for sure. Consequently, I'm not sure how or where to set it to fix this issue.

(Click on the function names in the above to go to the relevant calls in the code)

neurodroid, 2015-01-21:
The code you're referring to under src/libstfio/abf/axon was written by Axon Instruments, but my suspicion is that something's wrong with [how max_size is determined](https://github.com/neurodroid/stimfit/blob/b8326dbafa01ace1eb115ae84c1f9a236bd0fac1/src/libstfio/abf/abflib.cpp#L172) - at the moment it's using std::vector::max_size() to determine the maximal number of elements that a std::vector can hold, but on Windows it seems like we run out of memory long before max_size so that I've arbitrarily decided to divide max_size by 8. Maybe there's a smarter way of determining max_size, e.g. std::vector::capacity()?

kh90909, 2015-01-22:
I agree that there's something wrong with how `max_size` is determined, but I don't think that's the issue here. The problem is triggered by the code that handles when the gap-free file is larger than `max_size`.

I'm pretty sure this is because `ABF2_GetNumSamples()` returns the wrong value (zero) for the last episode, which seems due to `m_uLastEpiSize` never being set.

I think I've tracked down the problem. The ABF file open function should set `m_uLastEpiSize`, as indicated in a code comment [here](https://github.com/neurodroid/stimfit/blob/b8326dbafa01ace1eb115ae84c1f9a236bd0fac1/src/libstfio/abf/axon/AxAbfFio32/abffiles.cpp#L1666-1667), which the Axon-provided `ABF_ReadOpen()` does. However, libstfio implements its own open function which does not do this. I believe I've fixed the problem with commit kh90909/stimfit@59bca25. It works for my test file at least, but I notice that `ABF_ReadOpen()` also does some synch array stuff in addition to setting `m_uLastEpiSize`, for files of type `ABF_VARLENEVENTS` as well as `ABF_GAPFREEFILE`. It doesn't look like `stfio::importABF2File()` does this either, so it's possible related bugs might surface in future.

neurodroid, 2015-06-14:
Fix applied in d3f990ac

closed 2015-06-14 by neurodroid