neurodroid/stimfit #13: Program crashes when loading an ABF file

Rhodungeon, 2015-03-31:
Hello,

I'm running the latest version of Stimfit on Windows 7 64bit. The program crashes everytime I load a 6 MB ABF file. It's quite strange since I was able to load the same file using a previous version (0.13 for Windows 32bit).

Here I report the error message

Problem signature:
  Problem Event Name:	BEX64
  Application Name:	Stimfit.exe
  Application Version:	0.0.0.0
  Application Timestamp:	5507364d
  Fault Module Name:	MSVCR90.dll
  Fault Module Version:	9.0.30729.4940
  Fault Module Timestamp:	4ca2e32e
  Exception Offset:	00000000000552c0
  Exception Code:	c0000417
  Exception Data:	0000000000000000
  OS Version:	6.1.7601.2.1.0.256.1
  Locale ID:	1040
  Additional Information 1:	6564
  Additional Information 2:	65648a7e515df0f8c11af14e7a5024ee
  Additional Information 3:	8dc1
  Additional Information 4:	8dc1b00b347ae737189b882597daeb93

neurodroid, 2015-03-31:
Can you please send us the abf file?
c.schmidt-hieber_at_ucl_dot_ac_dot_uk
cc: alois.schloegl_at_ist_dot_ac_dot_at

neurodroid, 2015-03-31:
Try v0.14.8 that should fix this bug:
https://github.com/neurodroid/stimfit/releases/tag/v0.14.8windows
and let us know if it works.

kh90909, 2015-04-05:
I suspect this issue is another manifestation of #11, where `CFileDescriptor::m_uLastEpiSize` never gets set, and, as a result, the code always treats the length of the final section as zero.

I haven't tested the above commit, but from reading it, I'm reasonably confident that it only masks the underlying problem and will cause the final section to be always omitted. I believe the commit I mentioned in #11 (kh90909/stimfit@59bca25) should fix the underlying issue.

I believe the reason that this crash did not occur with 0.13 is that the libraries that version was built with do not include the `std::vector::operator[]` index out of range checks. Without these checks, the code silently ignores the final section of data rather than crashing.

neurodroid, 2015-04-05:
Yes you're right, your fix is better, applied in https://github.com/neurodroid/stimfit/commit/d3f990ac

neurodroid, 2015-04-05:
@Rhodungeon Please test v0.14.9 that includes another fix so that the end of the file isn't discarded:

https://github.com/neurodroid/stimfit/releases/tag/v0.14.9windows

Let us know if it works for you so that I can close this issue.

Rhodungeon, 2015-04-06:
Hi Christoph,

I've tested both the application and I can confirm they are able to load
abf files!

Thank you very much!

neurodroid, 2015-04-06:
@Rhodungeon Thanks for testing and for reporting the bug.

closed 2015-04-06 by neurodroid