neurodroid/stimfit #15: Unable to build under Arch linux

Taomach, 2015-04-05:
The problem is with wxwidgets. I have both 3.0 and 2.8 versions of wxgtk installed, so I run configure with  --enable-python and --with-wx-config=wx-config-2.8. After that, make fails with this log: https://gist.github.com/anonymous/02f587ae563e11edbd94 . How do I fix this?

neurodroid, 2015-04-05:
Explicitly indicating the Python 2.x binary during configure should fix this, e.g.:
```shell
PYTHON=/usr/bin/python2.7 ./configure ...
```

Taomach, 2015-04-05:
Thanks, but is it possible to build stimfit with python3?

neurodroid, 2015-04-05:
> Thanks, but is it possible to build stimfit with python3?

Unfortunately no. The main issue is that the wxPython API for the embedded shell hasn't been ported to Python 3 yet. I've raised this issue with the wxPython developers almost 2 years ago:

https://groups.google.com/forum/#!topic/wxpython-dev/2UmbQUr_6QM

If you'd like to build with Python 3, you should ask the wxPython developers to port the embedded sample to Phoenix (their codename for py3-wxPython)

Taomach, 2015-04-05:
Well, that's a pity. Specifying python version as you suggested earlier did not help too.

neurodroid, 2015-04-05:
Can you send or paste your `config.log` please?

Taomach, 2015-04-05:
https://gist.github.com/anonymous/30501b383c3ca10fb983

neurodroid, 2015-04-05:
Your `PYTHON` environment variable was ignored - see l.1167 of your `config.log`, that still points to your default system Python (`/usr/bin/python`, which is presumably python 3). Make sure that /usr/bin/python2.7 actually exists on your system. Also, note that the environment variable has to be set on the same line as the call to `configure`:
```shell
PYTHON=/usr/bin/python2.7 ./configure --enable-python
```
`--enable-ipython` hasn't been tested in a very long time - I doubt it will work.
If everything fails you'll have to explicitly set your PYTHON environment variable temporarily and then switch back to your default Python once you're done with `configure`:
```shell
TMPPYTHON=${PYTHON}
export PYTHON=/usr/bin/python2.7
./configure --enable-python
export PYTHON=${TMPPYTHON}
```
Make sure that in your config.log, `PYTHON` is set to python2.

neurodroid, 2015-04-06:
P.S. You can build the stfio module with Python 3

schloegl, 2024-03-22:
Stimfit can be build with python3 for quite some time. Assuming this is fixed.

closed 2024-03-22 by schloegl