neurodroid/stimfit #18: Convert to atf using stfio

Taomach, 2015-04-14:
Hello!
Stimfit has a dialog that allows converting files in different formats to Igor, Biosig of ATF, yet stfio documentation states that the only supported output format is HDF. Is there any way to use the ATF writer directly from stfio? I know how to construct ATF files by hand but I would very much appreciate the way to escape all that hassle.

schloegl, 2024-03-22:
Is still still a desired feature ? Would you mind explaining why you want to convert to ATF ? 
What kind of processing steps would like to do, that would require a conversion to ATF ?

schloegl, 2024-03-27:
Conversion to ATF is also supported by the [biosig-tools](https://biosig.sourceforge.net/download.html) with the command 
`    save2gdf -f=ATF infile  outfile.atf
`

Could you please check if this addresses thos issue ?

schloegl, 2025-07-09:
It seems there is no more information how users want use this feature, bosig's save2gdf and stimfit's save-as atf produce different files when the original data contains multiple sweeps and/or multiple channels. Without further user input, it is not clear what the expected behaviour is, or whether this feature is actually used and useful to anyone. Because further information is needed, and no further input is provided, I'm closing this ticket now.

closed 2025-07-09 by schloegl