neurodroid/stimfit #19: file conversion

bchagnaud, 2015-04-26:
Hi 
I am converting *.abf files with stimfit to *.ibw which works well. Thank you for that!!!!!
I would like to ask for one thing: could you make the window in which you search for the "to be converted files" (as well as the destination window) save the last chosen destination. That would be extremely helpfull. 
One other thing that would be great would be to: not have to care about subfolder when ones converts the files. At this point one has to click on each subfolder for the conversion to happen.

Thanks a lot!!!!
Best wishes
Boris

neurodroid, 2015-04-30:
@bchagnaud The first feature (restoring the most recent path) shouldn't be a problem at all. The second feature (recursing into subdirectories) is a bit more of a challenge. I hope I will find some time to implement these features this weekend.

neurodroid, 2015-05-04:
@bchagnaud Your suggestions have been implemented. Let me know which OS you use so that I can release a version for you to test.

bchagnaud, 2015-05-04:
Hi 
thank you so much.
I am using stimfit on a Windows PC.
Greets
Boris

neurodroid, 2015-05-04:
Try https://github.com/neurodroid/stimfit/releases/tag/v0.14.10windows
Bugs are likely at this stage

bchagnaud, 2015-05-07:
Hi
so I tried it, the folder selection is very nice, however suddenly the conversion to igor files has changed. It gives out a strange file name with the form: filename.ibw.tracename.ibw 
If you could fix that, that would be great ;-)
Greets
Boris

schloegl, 2024-03-22:
Only single channels  can be exported to IBW format; in order to address the issue for multichannel data, the channel name is included in the filename. You can always rename the files as needed. 

Moreover, I want to point out that you can export all data to HDF5, which is then can be imported to IgorPro 
(see https://www.wavemetrics.com/products/igorpro/dataaccess/fileformats) 

I assume these workrounds are good enough to address thes issue, and I'll close this.

closed 2024-03-22 by schloegl