neurodroid/stimfit #20: [Minor annoyance] Program requires a trace to b open to use the stf.new_window* commands

AndrewMicallef, 2015-06-13:
Hi,
I've come back to using stimfit for analysing binary data from an unsupported format. The format my data is in comes from a series of macros written for Igor a long time ago. Basically I'm dealing with one giant file of binary data, which has meta information stored as plain text in a log file (number of sweeps, sweep duration, sampling interval etc.).

I've written a parser, which reads the plain txt and then applies the relevant stimfit `set` commands. I then read the binary into a np array with a function that creates a 2d array (number of sweeps x sweep length).

I then have the parser put the np array into the stimfit gui window using `stf.new_window_matrix(np.array)`. Stimfit the responds to say there is no open file. If I open some unrelated file and then run my parser it all works, so my work flow now looks something like this:

1. open some file I don't care about
2. parse my archaic data
3. close the file I don't care about
4. have a coffee because 1-3 were such hard work

neurodroid, 2015-06-14:
It's a feature to encourage coffee consumption.
Which OS are you using?

AndrewMicallef, 2015-06-15:
My local cafe thinks this particular undocumented feature is great.

So I'm using Stimfit 0.14.10 on a windows 8 machine.

I updated the parser I'm using so that it automates the process of opening a blank file, processing my data, then switching to the blank file and closing it.

AndrewMicallef, 2015-06-16:
Although it would be kinda neat if I could get this sort of thing happening. Ie have python attempt to make a new window, if it fails, to the open up my dummy file and make the new window without bothering me with a pop-up window to tell me that it can't find an open file. 

``` <python>
#for completeness
v = np.random.rand(3,2)

try:
    stf.new_window_matrix(v)
except:
    stf.file_open(dummy_file)
```
At the moment the above code returns False if you run it in a new stimfit session
```
False
```

neurodroid, 2015-06-16:
05a00a30, mentioned above, should fix this issue, but I'm not sure when I will have time to build a new Windows version.

neurodroid, 2015-06-21:
Let me know if [v0.14.11](https://github.com/neurodroid/stimfit/releases/tag/v0.14.11windows) fixes your problem so that I can close this issue.

AndrewMicallef, 2015-06-23:
I can confirm that this works on Windows 8.
Thank you so much.

There is a coffee shop on campus that may go into liquidation now :)

neurodroid, 2015-06-23:
Don't worry - the program has some more undocumented "features" that will keep them in business.

closed 2015-06-23 by neurodroid