neurodroid/stimfit #22: Python interpreter History

AndrewMicallef, 2015-07-22:
Is there a way to get the history of all input to the stimfit python interpreter without copying all the output as well? I'm thinking of a functional equivalent of IPython's `%history` magic.

neurodroid, 2015-07-22:
Ideally for this we'd have to bring back ipython as the default shell. Otherwise we'd be reinventing the wheel. Unfortunately the wx version of ipython was abandoned a while ago and to my knowledge there are no plans to bring it back (Jose?).

JoseGuzman, 2015-07-29:
CheckIssue  #23 , we have been discussing about it since version of IPython 0.11. If I ever have the chance to have a sabbatical semester, I would work exclusively on that issue. I agree that an IPython-like terminal would fit so nice to Stimfit!