neurodroid/stimfit #24: Failed to build documentation

JoseGuzman, 2015-09-08:
When  running in $STIMFITHOME/doc/sphinx/ the following command line:
$> sphinx-build ./ ~/tmp/stf/doc

I've got the following error

Running Sphinx v1.1.3
loading pickled environment... done
No builder selected, using default: html

Theme error:
no theme named 'alabaster' found (missing theme.conf?)

neurodroid, 2015-09-08:
You can get it from pip, e.g.
`sudo pip-3.4 install alabaster`
or whatever version of pip you want to use

JoseGuzman, 2015-09-08:
the following line should be added to conf.py

html_theme_path = ["/usr/local/lib/python2.7/dist-packages/"]

I will submit a pull request as soon as I learn how to do it :+1:

neurodroid, 2015-09-08:
I don't think this should be added as it will look different depending on which platform and which Python version you use. I don't have a single system where this path would be correct.

JoseGuzman, 2015-09-09:
I think the easiest way to go with this is suggested in Pull request #25. Alternativerly, in the future, we may include a custom-made theme within the repository.

neurodroid, 2015-09-09:
OK, merged #25, closing this issue. Thanks for your pr.

closed 2015-09-09 by neurodroid