neurodroid/stimfit #27: stf.set_recording_time misbehaving

AndrewMicallef, 2015-11-17:
Hi,

I just encountered this bug when trying to construct a recording object and to set the recording time.

```python
In [1]: stf.set_recording_time('16:45:10')
Out[1]: True
In[2]: stf.get_recording_time()
Out[2]: '16:15:26'
```
Basically the minutes and the seconds aren't being set correctly.

I get the same behavior when using `rec.date = '16:45:10'`

neurodroid, 2015-11-17:
Can you please test
```python
stf.set_recording_time('16-45-10')
```

neurodroid, 2015-11-17:
@schloegl This looks like a problem with [Recording::SetTime()](https://github.com/neurodroid/stimfit/blob/master/src/libstfio/recording.cpp#L128)
Either sscanf only uses the first format string regardless of whether there are dashes or not, or alternatively strptime fails. I think you wrote this function (correct me if I'm wrong). Could you please look into this?

schloegl, 2025-07-09:
I checked this with the lastest version of stimfit 0.16.7 (in Debian13/trixie) as well as compiled from source, and get 

```
>>> stf.set_recording_time('16:45:10')
True
>>> stf.get_recording_time()
'16:45:10'
```

So this issue seems to be solved - at least in recent versions of stimfit on Linux.

closed 2025-07-09 by schloegl