neurodroid/stimfit #28: Problems with toolbars in MacOSX 10.10.5

fjohenning, 2015-12-01:
The software starts and i can use most of the functions. However, when I select traces for averaging, the analysis  toolbar only works on the first window opened after starting stimfit from scratch. when I have several windows open, it seems stimfit does not recognize which selection is the most recent. In addition, the navigation toolbar does not respond at all.
I hope this makes sense, and there is a solution, as I really started to like working with stimfit.

neurodroid, 2015-12-01:
This bug should partly be fixed now. It might take a while until the new version hits the MacPorts repositories (see https://trac.macports.org/ticket/49873 for updates). Please let me know if any remaining issues.

fjohenning, 2015-12-02:
Great, thank you very much!
As I am new to using macport for installation: how do I update my stimfit from terminal using macport. Sorry for such a  naive question,
Best
Friedrich

PD Dr. Friedrich Johenning
Charité - Universitätsmedizin Berlin
CharitéCrossover - Neurowiss. Forschungszentrum
AG Schmitz
Ebene 03 Räume 101 - 116
Virchowweg 6 - Charitéplatz 1
10117 Berlin

phone: ++49 30 450639064
fax: ++49 30 450539943

neurodroid, 2015-12-02:
Once it's uploaded to the MacPorts repository:
```bash
sudo port selfupdate
sudo port upgrade -u outdated
```

neurodroid, 2015-12-02:
The update should be available now.

neurodroid, 2015-12-09:
@fjohenning Please let me know if the latest updates have fixed this issue so that we can close it

fjohenning, 2015-12-09:
Hi,,
the toolbars also work. thank you very much.
Best Friedrich

closed 2015-12-09 by neurodroid