neurodroid/stimfit #29: Problem unselecting false-positive traces in MacOS X 10.10.5

fjohenning, 2015-12-03:
Hi,
working with my data, I encountered another problem: When I have the events detected and want to go through the single events in the edit mode, I can add new events (false negatives), but clicking the checkboxes next to the arrows, it is not possible to get rid of the false positives as indicated in the manual.
Best regards
Friedrich

neurodroid, 2015-12-05:
Fixed in 0.14.13, pending upload to MacPorts (https://trac.macports.org/ticket/49914)

neurodroid, 2015-12-08:
@fjohenning An updated version should now be available via MacPorts:
```bash
sudo port selfupdate
sudo port upgrade -u outdated
```
Let me know if this works for you so that we can close this issue

fjohenning, 2015-12-08:
Thank you very much!
The update works great.

closed 2015-12-09 by neurodroid