neurodroid/stimfit #31: unique recording time for each trace

AndrewMicallef, 2015-12-15:
Hi, 

In my current set of experiments I am acquiring data at random intervals using an external trigger to start the recordings.

Because of this I would like to include the unique start time of each trace in the stimfit file (which I generate from a [wavesurfer](https://github.com/JaneliaSciComp/Wavesurfer) file.)

Basically is this a feature that currently exists in the stimfit recording object and if not do you have any suggestions on a hack?

Also in case you are interested this is the script I am using to make the conversions:
https://gist.github.com/AndrewMicallef/0d3793590f01a7daa88b

- Regards Andrew

neurodroid, 2015-12-15:
Supposedly setting recording.date to a [datetime.datetime](https://docs.python.org/2/library/datetime.html) object doesn't work?

schloegl, 2018-02-07:
The start time per trace/sweep is currrently not part of the core object of stimfit, also it could be added if necessary. The question is why or how should stimfit make use of this information ? Can you explain the purpose and possible benefit of this for users of stimfit ?