neurodroid/stimfit #32: Problem reading HEKA PatchMaster .dat fles

Taomach, 2015-12-25:
I am using standalone stfio module to read data files. Whenever I try to read any HEKA .dat file, the resulting object is empty. If I specify the file type using `ftype='heka'` argument, I get `StfIOException: 'Error reading file'`.

[Example file.zip](https://github.com/neurodroid/stimfit/files/72206/Example.file.zip)

neurodroid, 2015-12-25:
@schloegl Can you help with this?

schloegl, 2017-07-06:
This is a late reply, in case it is not solved yet, one needs to build stimfit with biosig support (configure --with-biosig or  --with-biosig2 (--with-biosiglite most likely will also do), than the support for HEKA v2 file format should be available. The file format should be automatically recognised. 

If not, please provide a file that shows the problem.

schloegl, 2018-02-07:
The file is Example_file.zip is a Heka file with an empty or invalid bundle header (a.k.a. HEKA v1, a.k.a.  HEKA DAT1). Currently, only HEKA files with valid bundle header (a.k.a.  HEKA v2, a.k.a. HEKA DAT2) are supported. 

As a workaround can open the file again with Patchmaster and save it with the option "Make Bundle File" (see Patchmaster Reference Manual); this will store the file in the "bundled heka format" (a.k.a. HEKA DAT2)

There are no immediate plans to add support plans to fix this.

closed 2018-02-07 by schloegl