neurodroid/stimfit #33: Data window fails to move in the edit mode after manually adding/deleting an event

niksGIT, 2016-01-16:
Hi there,

using the 0.14.13 (64 bit) beta release for Windows I encountered the following problem: Once an event is manually added or deleted in the editing mode, the convenient shift(or CTRL)+arrow key-triggered movement of the data window is not functional any more.
It would be extremely helpful if you could sort this out, as the use of the arrow buttons (in the context menu) to shift the window is quite tedious... 
Thanks so much!
-Nikolaus

neurodroid, 2016-01-17:
@niksGIT I don't have access to a Windows system right now. On Mac and Linux, while in event mode, if I left click somewhere into the window that shows the trace ("data window") after adding or deleting an event, I can move the trace around with the keyboard again.
Does this work on Windows as well?

niksGIT, 2016-01-17:
... unfortunately, this is exactly what is not working (I just tried again).

niksGIT, 2016-01-19:
I have to specify: In most cases Adding events does work, while the Deletion of an event always leads to the described freezing of the window (moving the trace with ctrl/or shift+arrow keys is blocked). Maybe this helps to narrow down the bug...

niksGIT, 2016-01-19:
OK, I found a way to 'unfreeze' the trace: when this happens expanding the time axis by clicking the 'magnifying glass' button in the context menu helps to enable the trace moving again. This time-axis expansion has to be done several times in a row -- clicking the button 10x in a row usually works.

neurodroid, 2016-01-20:
@niksGIT [Test 0.14.14](https://github.com/neurodroid/stimfit/releases/tag/v0.14.14windows). A simple left-click should now return focus to the data window so that it can capture keyboard input.

neurodroid, 2016-02-02:
Closing this issue as I haven't heard back from @niksGIT in 2 weeks

closed 2016-02-02 by neurodroid