neurodroid/stimfit #35: Build on ubuntu fails

aolsux, 2016-02-15:
Hi,
I have tried the packetmanager installation for stimfit on my ubuntu machine. Installing the package worked, but running stimfit resultet in an errormessage "cannot import wxversion".
Hence I tried to build it from source (I think I have some experience doing such things...). I followed the instructions from [here](http://www.stimfit.org/doc/sphinx/linux_install_guide/requirement.html). Autogen and configure did well. Compiling also worked. However linking fails due to missing libtiff4 libraries (`/usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-2.8.so: undefined reference to TIFFSetDirectory@LIBTIFF_4.0`). I have done a little googling, and learned, that the newer ubuntus ship libtiff5. It seems that there is no (easy)  way to install libtiff4 via apt-get :-(
I have attached my [config.log](https://github.com/neurodroid/stimfit/files/130509/config.log.txt) for further information. 
I have no idea whether its a straight forward thing to fix, but I just wanted you to know :-D
HtH Martin

aolsux, 2016-02-15:
Do you think installing wxWidgets 2.9/3.0 can improve the situation?

aolsux, 2016-02-15:
After another day in AutoHell, I did not succeed to make stimfit compile on my machine. I updated to `xubuntu 15.10`, which comes with `wxwidgets3.x`, still the linker tries to link to `ibtiff4`, while `ldd` says wxwidgets depends on libtiff5. :-( Maybe its an issue with the libtool that comes with stimfit, or it might just be some messed up system configuration on my side. I cant tell :-/ If you want more information about the issue, let me know. Anyhow, maybe you have considered switching your build system to CMake, since you want to provide multi platform support (including windows) i can only recommend that :P
HtH Martin

neurodroid, 2016-02-18:
@aolsux I can look into this when I'm back at work next week

schloegl, 2018-02-07:
@aolsux: did you manage to solve the build problem of Stimfit on Ubuntu ? 

If not, make sure that all wx-2.8 packages are removed and install wx-3.0 packages only. 

If it does not work for you, please send the detailed error message and the output of 
 `  lsb_release -a
`

aolsux, 2018-02-07:
Hi,
I am unsure if I was able to solve it. Since I have changed work/topic I will also not find time to retest/retry - Sorry. I will just close the issue. Maybe it will help someone who has the same problem, then it can be reopened...
Regards,
Martin

closed 2018-02-07 by aolsux