neurodroid/stimfit #39: Custom hotkeys / key binding for changing channels

AndrewMicallef, 2016-04-05:
The manual doesn't mention anything about being able to swap channels from the key board. I know I could type `stf.set_channel(N)` each time, but it would be a nice enhancement to be able to do something similar to the amazing keyboard scrolling you have for flipping through traces. Perhaps use `Shift + cursor key` ?

JoseGuzman, 2016-04-05:
Simply press the X key to swap channels.  I'll add it to the online manual

AndrewMicallef, 2016-04-05:
Thanks, X works well for two channels, although for files with more than two X only toggles between reference and active. Basically the use case is I have files with N channels (usually 5 but can be more) and I want to scroll through the channels.

JoseGuzman, 2016-04-05:
I tend to use the “Channel section” menu for moving along ADC channels when having 8 ADC. I very rarely use `stf.set_channel(N)`. Would you recommend to have short-key of the type Shift+1, Shift+2, etc... to move around the channels?
@neurodroid, any suggestion?

AndrewMicallef, 2016-04-05:
Either that, or to be consistant with the current bindings perhaps ctrl+up key / ctrl+down key, seeing as left and right scroll through traces, it would make sense that up and down move through the other relevant dimension

neurodroid, 2016-04-05:
> Also, I'd be happy to make the change for my own copy, if you could point me to the file that would need editing...

https://github.com/neurodroid/stimfit/blob/master/src/stimfit/gui/graph.cpp#L1339

Please send a pull request once you've added the new key bindings. Thanks!

AndrewMicallef, 2016-05-19:
I had a go partially implementing the feature I want. I basically just copied the set_channel function into the file you pointed me to. I haven't had a chance to test this yet as i haven't got all the dependancies on my Windows machine, and building on ubuntu seems to be bugging out.

this is the output I get from make, on the master branch:
```
make  all-recursive
make[1]: Entering directory `/home/andrew/stimfit'
Making all in src
make[2]: Entering directory `/home/andrew/stimfit/src'
Making all in libstfio
make[3]: Entering directory `/home/andrew/stimfit/src/libstfio'
make[3]: Nothing to be done for `all'.
make[3]: Leaving directory `/home/andrew/stimfit/src/libstfio'
Making all in libstfnum
make[3]: Entering directory `/home/andrew/stimfit/src/libstfnum'
make[3]: Nothing to be done for `all'.
make[3]: Leaving directory `/home/andrew/stimfit/src/libstfnum'
Making all in stimfit
make[3]: Entering directory `/home/andrew/stimfit/src/stimfit'
  CXX      libstimfit_la-app.lo
In file included from ./gui/app.cpp:50:0:
./gui/../../../stfconf.h:63:17: error: expected unqualified-id before string constant
 #define VERSION "0.15.2"
                 ^
make[3]: *** [libstimfit_la-app.lo] Error 1
make[3]: Leaving directory `/home/andrew/stimfit/src/stimfit'
make[2]: *** [all-recursive] Error 1
make[2]: Leaving directory `/home/andrew/stimfit/src'
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory `/home/andrew/stimfit'
make: *** [all] Error 2
```
Not sure if this relates to: #35 ?

[(https://github.com/AndrewMicallef/stimfit/tree/channel_scroll)]

AndrewMicallef, 2016-05-19:
Also is there already a variable that holds the number of channels in a document. Or a neat way of accessing this? I want the scrolling in the negative direction to rollover in the same way that scrolling through traces works at the moment.

neurodroid, 2016-05-19:
You're probably building in tree. It's better to make a separate build directory, e.g.
```bash
cd ~/stimfit
mkdir build
cd build
../configure
make
```
You need to re-generate the original stfconf.h file in the top-level directory, e.g. by doing
```bash
cd ~/stimfit
rm stfconf.h
git checkout stfconf.h
```
Looks like the stfconf.h file that gets generated by `configure` is not usable - we'll have to fix this at some point.

AndrewMicallef, 2016-05-19:
Cool, I'll have a go at that. I just spent the evening trying to get all the dependencies together on Debian with partial success. I have Debian at home, and ubuntu at work, not the smartest setup.

Long story short it'll probably be a couple more days till I can build and test my efforts.

neurodroid, 2016-05-19:
```bash
sudo apt-get build-dep stimfit
```
should ideally get you all build dependencies

AndrewMicallef, 2016-05-19:
Also, as an aside, is there a way to set `nogimicks` in stimfit itself? It would sometimes be nice to see just the raw data without all the overlaid cursors.

AndrewMicallef, 2016-05-20:
No luck building master branch on Ubuntu. I ran `sudo apt-get build-dep stimfit` on this machine just before attempting to build... This is not on the branch that I made my modifications on, I want to check I can build the original before testing if mine works.
```
andrew@andrew-VirtualBox:~/stimfit/build$ ../configure
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking for style of include used by make... GNU
checking dependency style of gcc... gcc3
checking build system type... x86_64-unknown-linux-gnu
checking host system type... x86_64-unknown-linux-gnu
checking how to print strings... printf
checking for a sed that does not truncate output... /bin/sed
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for fgrep... /bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking whether the shell understands some XSI constructs... yes
checking whether the shell understands "+="... yes
checking how to convert x86_64-unknown-linux-gnu file names to x86_64-unknown-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-unknown-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for mt... mt
checking if mt is a manifest tool... no
checking how to run the C preprocessor... gcc -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for dlfcn.h... yes
checking for objdir... .libs
checking if gcc supports -fno-rtti -fno-exceptions... no
checking for gcc option to produce PIC... -fPIC -DPIC
checking if gcc PIC flag -fPIC -DPIC works... yes
checking if gcc static flag -static works... yes
[... 101 lines cut ...]
                 from ../../../src/stimfit/./gui/./app.h:176,
                 from ../../../src/stimfit/./gui/doc.cpp:35:
../../../src/stimfit/./gui/././../../libstfio/./recording.h: In copy constructor 'Recording::Recording(const Recording&)':
../../../src/stimfit/./gui/././../../libstfio/./recording.h:41:16: warning: 'Recording::time0' is deprecated (declared at ../../../src/stimfit/./gui/././../../libstfio/./recording.h:378) [-Wdeprecated-declarations]
 class StfioDll Recording {
                ^
../../../src/stimfit/./gui/././../../libstfio/./recording.h:41:16: warning: 'Recording::time0' is deprecated (declared at ../../../src/stimfit/./gui/././../../libstfio/./recording.h:378) [-Wdeprecated-declarations]

                 from /usr/lib/python2.7/dist-packages/numpy/core/include/numpy/arrayobject.h:4,
                 from ../../../../src/stimfit/py/pystf_wrap.cxx:3062:
/usr/lib/python2.7/dist-packages/numpy/core/include/numpy/npy_1_7_deprecated_api.h:15:2: warning: #warning "Using deprecated NumPy API, disable it by " "#defining NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]
 #warning "Using deprecated NumPy API, disable it by " \
  ^
  CXXLD    libpystf.la
make[3]: Leaving directory `/home/andrew/stimfit/build/src/stimfit/py'
make[3]: Entering directory `/home/andrew/stimfit/build/src'
make[3]: Nothing to be done for `all-am'.
make[3]: Leaving directory `/home/andrew/stimfit/build/src'
make[2]: Leaving directory `/home/andrew/stimfit/build/src'
make[2]: Entering directory `/home/andrew/stimfit/build'
  CXX      stimfit-main.o
  CXXLD    stimfit
make[2]: Leaving directory `/home/andrew/stimfit/build'
make[1]: Leaving directory `/home/andrew/stimfit/build'
```

AndrewMicallef, 2016-05-20:
actually I just tried `sudo make` and this was the output.

```
andrew@andrew-VirtualBox:~/stimfit/build$ sudo make
make  all-recursive
make[1]: Entering directory `/home/andrew/stimfit/build'
Making all in src
make[2]: Entering directory `/home/andrew/stimfit/build/src'
Making all in libstfio
make[3]: Entering directory `/home/andrew/stimfit/build/src/libstfio'
make[3]: Nothing to be done for `all'.
make[3]: Leaving directory `/home/andrew/stimfit/build/src/libstfio'
Making all in libstfnum
make[3]: Entering directory `/home/andrew/stimfit/build/src/libstfnum'
make[3]: Nothing to be done for `all'.
make[3]: Leaving directory `/home/andrew/stimfit/build/src/libstfnum'
Making all in stimfit
make[3]: Entering directory `/home/andrew/stimfit/build/src/stimfit'
make[3]: Nothing to be done for `all'.
make[3]: Leaving directory `/home/andrew/stimfit/build/src/stimfit'
Making all in stimfit/py
make[3]: Entering directory `/home/andrew/stimfit/build/src/stimfit/py'
make[3]: Nothing to be done for `all'.
make[3]: Leaving directory `/home/andrew/stimfit/build/src/stimfit/py'
make[3]: Entering directory `/home/andrew/stimfit/build/src'
make[3]: Nothing to be done for `all-am'.
make[3]: Leaving directory `/home/andrew/stimfit/build/src'
make[2]: Leaving directory `/home/andrew/stimfit/build/src'
make[2]: Entering directory `/home/andrew/stimfit/build'
make[2]: Leaving directory `/home/andrew/stimfit/build'
make[1]: Leaving directory `/home/andrew/stimfit/build'
```

but then when I try to run stimfit:

```
andrew@andrew-VirtualBox:~/stimfit/build$ ./stimfit
collect2: error: ld returned 1 exit statusnandrew@andrew-VirtualBox:~/stimfit/build$ 
```

neurodroid, 2016-05-20:
@AndrewMicallef Looks like `make` works fine on your system. There are a couple of warnings but no errors. To get everything into place and linked correctly you have to run `sudo make install` at the end, which will typically install stimfit into `/usr/local/bin/stimfit`

AndrewMicallef, 2016-09-09:
Hey, Sorry for the slow progress on this, I have had other things to do.

I almost have it working. I managed to build no troubles, by following your instructions and made the two functions I needed to for scrolling through channels.

The fork is here, not sure how to make a pull request. https://github.com/AndrewMicallef/stimfit/tree/channel_scroll

It almost works, but scrolling up throws an error when you get to the first channel. Ie, i can move from the last channel to the first, but going from the first to the next gives me an error. I'm not sure how to catch this error, and how to handle it.

the error message is 

```
Error while changing channels
Please close file
channel out of range in Recording::SetSecChIndex()
```

AndrewMicallef, 2016-09-20:
I worked out what a pull request means, and made a new one #45. I adjusted the code since last week and think it works better now. The scrolling only affects the active channel, leaving reference channel in place. Which works better with `X` now that I think about it too, because you can scroll through the channels, then hit `X` to set your reference.

Also rollover works in both directions...

neurodroid, 2016-09-25:
Closed by f5058524

closed 2016-09-25 by neurodroid

AndrewMicallef, 2016-09-28:
Any chance of a new windows binary? (I don't have admin access on a work computer to roll my own)