neurodroid/stimfit #51: Abf batch converstion to Atf fails when abf has two signals

metracy, 2016-11-22:
When I use the abf to atf conversion (so I can use R with the files), I run into an issue with only one signal within the abf being converted... I.e. I have ImemB and ImemC. Is there a way to extract only the imemC in the batch file conversion?

neurodroid, 2016-11-22:
@JoseGuzman Any ideas? I haven't looked into the batch conversion in a while...

metracy, 2016-11-22:
I can provide an example abf file it that helps. Also, how difficult would it be to run the batch process a python command?

Taomach, 2016-11-23:
@metracy, you can use the standalone stfio python module to import data from ABF, and then export it with numpy. It is as simple as this:
`import numpy as np
import stfio
import sys
import os

for abf_file in sys.argv[1:]:
    data = stfio.read(abf_file)
    si = data.dt
    points = len(data[0][0])
    times = np.arange(0, points * si, si)
    signum = 0
    for signal in data:
        arr = np.vstack((times, signal))
        file_name = os.path.splitext(abf_file)[0] + '_signal' + str(signum) + '.csv'
        print(file_name)
        np.savetxt(file_name, arr.T)
        signum += 1`

neurodroid, 2016-11-23:
Thanks @Taomach !

metracy, 2016-11-23:
wow @Taomach thanks!

Could you point me to a simple way analyze frequency spikes over a series of sweeps from wholecell patch clamp data? I haven't been able to understand much of the spike train analysis packages used in R or python.

Taomach, 2016-11-23:
@metracy, I have no experience with spike train analysis. I think, the easiest way would be to implement a simple threshold event detection, and then find the frequency as a inverse of inter-spike intervals. You can also try `scipy.signal.find_peaks_cwt`, although I found it quite finicky on electrophysiological data. Alternatively, you can implement some more complex event detection algorythm (good example: http://www.cell.com/biophysj/fulltext/S0006-3495(12)00935-6), but for spike trains it may be an overkill.

metracy, 2016-12-02:
@Taomach 

I modified the code as follows to create a dataframe ready csv for current clamp data with twenty sweeps.

`#!/usr/bin/python
import numpy as np
import stfio
import os
import glob

abf_list = glob.glob('./*.abf')
for abf_file in abf_list:
    data = stfio.read(abf_file)
    si = data.dt
    points = len(data[0][0])
    times = np.arange(0, points * si, si)
    for signal in data:
        arr = np.vstack((times, signal))
        file_name = os.path.splitext(abf_file)[0] + '_' + signal.name + '.csv'
        np.savetxt(file_name, arr.T,delimiter=',', header="time,sweep_1,sweep_2,sweep_3,sweep_4,sweep_5,sweep_6,sweep_7,sweep_8,sweep_9,sweep_10,sweep_11,sweep_12,sweep_13,sweep_14,sweep_15,sweep_16,sweep_17,sweep_18,sweep_19,sweep_20", comments='')
`

metracy, 2016-12-02:
not sure why code wouldn't format in comment

schloegl, 2024-03-22:
It seems the suggestion addressed the specific issue of @metracy.

Before closing this issue, I want to point put that save2gdf from the [biosig-tools](https://biosig.sourceforge.net/download.html) provides a number of data convers routines. Converting ABF data into some CSV or ASCII format could be done with these commands: 
```
save2gdf  -CSV yourfile.abf  > yourfile.csv
save2gdf  -f=ASCII yourfile.abf yourfile.asc
```
Many other conversions are supported, you can see this with 
```
save2gdf  --help
```

Biosig has also some  language binding of libbiosig to R, which might be of interest to @metracy.

closed 2024-03-22 by schloegl