neurodroid/stimfit #52: narrowing build failure

iainlane, 2016-11-30:
With this gcc version

```
root@ethical-kitten:~/stimfit-0.15.3# g++ --version
g++ (Ubuntu/Linaro 6.2.1-5ubuntu1) 6.2.1 20161124
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

stimfit 0.15.3 fails to build on Ubuntu armhf like so:

```
CXX      libbiosiglite_la-scp-decode.lo
In file included from biosig4c++/t210/scp-decode.cpp:113:0:
biosig4c++/t210/codes.h:25:37: error: narrowing conversion of ‘-1’ from ‘int’ to ‘char’ inside { } [-Wnarrowing]
 static const char STR_END[]={-1,'\0'};
```

I tried making `STR_END` be a `signed char`, but this requires a lot of changes.

As a workaround I'll build with `-Wno-narrowing`.

neurodroid, 2016-12-10:
Thanks for reporting. Also addresses Debian bug [#847526](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=847526)

neurodroid, 2016-12-10:
Fixed [upstream](https://sourceforge.net/p/biosig/code/ci/8e7c3d7e)

commit c1ed1b09 (2016-12-10): Fix compilation with gcc6 on armhf. Closes #52
closed 2016-12-10 by neurodroid