neurodroid/stimfit #54: Reading CFS files

btyukodi, 2017-01-09:
I am trying to run Stimfit on Ubuntu 16.04. Installation went without any issues with 

    sudo apt-get install stimfit python-stfio

Then I tried to open the attached CFS (and many other) file and it failed with segmentation fault. The file opens up with no issues with Signal 4.06. Any ideas?

[seal test.cfs.zip](https://github.com/neurodroid/stimfit/files/693636/seal.test.cfs.zip)

The full traceback is below:

./stfio.cpp (line 155): importABF attempted
*** Error in `stimfit': double free or corruption (!prev): 0x00000000039de370 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7f618ea6b7e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x7fe0a)[0x7f618ea73e0a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7f618ea7798c]
/lib/x86_64-linux-gnu/libc.so.6(fclose+0x103)[0x7f618ea61363]
/usr/lib/stimfit/libstfio.so(c_CloseHandle+0x9)[0x7f618a9fa9d9]
/usr/lib/stimfit/libstfio.so(_Z14CloseHandleBufP12ATF_FILEINFO+0x79)[0x7f618aa00559]
/usr/lib/stimfit/libstfio.so(ATF_OpenFile+0x240)[0x7f618a9fe440]
/usr/lib/stimfit/libstfio.so(_ZN5stfio13exportATFFileERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERK9Recording+0xe7)[0x7f618aa04a97]
/usr/lib/stimfit/libstimfit.so(_ZN16wxStfParentFrame9OnConvertER14wxCommandEvent+0x1001)[0x7f618f432fe1]
/usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0(_ZNK16wxAppConsoleBase16CallEventHandlerEP12wxEvtHandlerR14wxEventFunctorR7wxEvent+0x3e)[0x7f618ff951fe]
/usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0(_ZN12wxEvtHandler23ProcessEventIfMatchesIdERK21wxEventTableEntryBasePS_R7wxEvent+0x57)[0x7f619011b6c7]
/usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0(_ZN16wxEventHashTable11HandleEventER7wxEventP12wxEvtHandler+0x8b)[0x7f619011b7bb]
/usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0(_ZN12wxEvtHandler11TryHereOnlyER7wxEvent+0x4b)[0x7f619011bb6b]
/usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0(_ZN12wxEvtHandler10DoTryChainER7wxEvent+0x43)[0x7f619011b973]
/usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0(_ZN12wxEvtHandler12ProcessEventER7wxEvent+0x45)[0x7f619011bc55]
/usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0(_ZN12wxWindowBase8TryAfterER7wxEvent+0x68)[0x7f618fb37d08]
/usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0(_ZN12wxEvtHandler18SafelyProcessEventER7wxEvent+0x7)[0x7f619011b9c7]
/usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0(_ZN10wxMenuBase9SendEventEii+0x9f)[0x7f618fadcb1f]
/usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0(+0x3040eb)[0x7f618f9c00eb]
/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(g_closure_invoke+0x145)[0x7f618cbddfa5]
/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(+0x21fc1)[0x7f618cbeffc1]
/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(g_signal_emit_valist+0xfbc)[0x7f618cbf8d5c]
/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(g_signal_emit+0x8f)[0x7f618cbf908f]
/usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0(gtk_widget_activate+0x6e)[0x7f618dcae67e]
/usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0(gtk_menu_shell_activate_item+0xfd)[0x7f618dba98ed]
/usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0(+0x144c56)[0x7f618dba9c56]
/usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0(+0x132afc)[0x7f618db97afc]
/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(g_closure_invoke+0x145)[0x7f618cbddfa5]
/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(+0x2256e)[0x7f618cbf056e]
/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(g_signal_emit_valist+0xa59)[0x7f618cbf87f9]
/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(g_signal_emit+0x8f)[0x7f618cbf908f]
/usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0(+0x24a8cc)[0x7f618dcaf8cc]
/usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0(gtk_propagate_event+0xc4)[0x7f618db96294]
/usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0(gtk_main_do_event+0x2cb)[0x7f618db9664b]
/usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0(+0x5ac4c)[0x7f618d80ac4c]
/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x2a7)[0x7f618c9071a7]
/lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x4a400)[0x7f618c907400]
/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_loop_run+0xc2)[0x7f618c907722]
/usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0(gtk_main+0xb7)[0x7f618db956a7]
/usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0(_ZN14wxGUIEventLoop5DoRunEv+0x25)[0x7f618f922ed5]
/usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0(_ZN15wxEventLoopBase3RunEv+0xa3)[0x7f618ffd7343]
/usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0(_ZN16wxAppConsoleBase8MainLoopEv+0x56)[0x7f618ff9c666]
/usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0(_Z7wxEntryRiPPw+0x70)[0x7f6190028f50]
stimfit[0x407052]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0)[0x7f618ea14830]
stimfit[0x4070c9]
======= Memory map: ========
00400000-0040a000 r-xp 00000000 08:05 279687                             /usr/bin/stimfit
00609000-0060a000 r--p 00009000 08:05 279687                             /usr/bin/stimfit
0060a000-0060b000 rw-p 0000a000 08:05 279687                             /usr/bin/stimfit
02219000-03bed000 rw-p 00000000 00:00 0                                  [heap]
7f6154000000-7f6154021000 rw-p 00000000 00:00 0 
7f6154021000-7f6158000000 ---p 00000000 00:00 0 
7f615b9bf000-7f615b9ca000 r-xp 00000000 08:05 1200050                    /lib/x86_64-linux-gnu/libnss_nis-2.23.so
7f615b9ca000-7f615bbc9000 ---p 0000b000 08:05 1200050                    /lib/x86_64-linux-gnu/libnss_nis-2.23.so
7f615bbc9000-7f615bbca000 r--p 0000a000 08:05 1200050                    /lib/x86_64-linux-gnu/libnss_nis-2.23.so
7f615bbca000-7f615bbcb000 rw-p 0000b000 08:05 1200050                    /lib/x86_64-linux-gnu/libnss_nis-2.23.so
[... 209 lines cut ...]
7f6164e67000-7f6166e67000 rw-p 00000000 00:00 0 
7f6166e67000-7f616ae67000 rw-p 00000000 00:00 0 
7f616ae67000-7f616ae68000 ---p 00000000 00:00 0 
7f616ae68000-7f616b668000 rw-p 00000000 00:00 0 
7f616b668000-7f616b669000 ---p 00000000 00:00 0 
7f616b669000-7f616be69000 rw-p 00000000 00:00 0 
7f616be69000-7f616be6a000 ---p 00000000 00:00 0 
7f616be6a000-7f616c66a000 rw-p 00000000 00:00 0 
7f616c66a000-7f616c75a000 r-xp 00000000 08:05 1333521                    /usr/local/lib/python2.7/dist-packages/numpy/.libs/libgfortran-ed201abd.so.3.0.0
7f616c75a000-7f616c959000 ---p 000f0000 08:05 1333521                    /usr/local/lib/python2.7/dist-packages/numpy/.libs/libgfortran-ed201abd.so.3.0.0
7f616c959000-7f616c95b000 rw-p 000ef000 08:05 1333521                    /usr/local/lib/python2.7/dist-packages/numpy/.libs/libgfortran-ed201abd.so.3.0.0
7f616c95b000-7f616c95c000 rw-p 00000000 00:00 0 
7f616c95c000-7f616c963000 rw-p 000f2000 08:05 1333521                    /usr/local/lib/python2.7/dist-packages/numpy/.libs/libgfortran-ed201abd.so.3.0.0
7f616c963000-7f616ec77000 r-xp 00000000 08:05 1333522                    /usr/local/lib/python2.7/dist-packages/numpy/.libs/libopenblasp-r0-39a31c03.2.18.so
7f616ec77000-7f616ee76000 ---p 02314000 08:05 1333522                    /usr/local/lib/python2.7/dist-packages/numpy/.libs/libopenblasp-r0-39a31c03.2.18.so
7f616ee76000-7f616ee95000 rw-p 02313000 08:05 1333522                    /usr/local/lib/python2.7/dist-packages/numpy/.libs/libopenblasp-r0-39a31c03.2.18.so
7f616ee95000-7f616eef8000 rw-p 00000000 00:00 0 
7f616eef8000-7f616ef74000 rw-p 02425000 08:05 1333522                    /usr/local/lib/python2.7/dist-packages/numpy/.libs/libopenblasp-r0-39a31c03.2.18.so
7f616ef74000-7f616f13d000 r-xp 00000000 08:05 1332170                    /usr/local/lib/python2.7/dist-packages/numpy/core/multiarray.so
7f616f13d000-7f616f33d000 ---p 001c9000 08:05 1332170                    /usr/local/lib/python2.7/dist-packages/numpy/core/multiarray.so
7f616f33d000-7f616f34c000 rw-p 001c9000 08:05 1332170                    /usr/local/lib/python2.7/dist-packages/numpy/core/multiarray.so
7f616f34c000-7f616f35e000 rw-p 00000000 00:00 0 
7f616f35e000-7f616f364000 rw-p 00902000 08:05 1332170                    /usr/local/lib/python2.7/dist-packages/numpy/core/multiarray.so
7f616f364000-7f616f3e4000 rw-p 00000000 00:00 0 
7f616f3e4000-7f616f4fb000 r-xp 00000000 08:05 547881                     /usr/lib/python2.7/dist-packages/wx-3.0-gtk2/wx/_misc_.x86_64-linux-gnu.soAborted (core dumped)

neurodroid, 2017-01-09:
I could reproduce the crash with the current version from the ubuntu repositories.
The crash is fixed in the latest version of the code if you `configure --without-biosig --enable-module`
Let me know if this works for you.
You could do me a favour and reply to this [Debian bug report](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=847526) asking that the pending package be uploaded from Debian mentors so that the version in the ubuntu and Debian repositories gets updated. I've already reminded them but haven't heard back.

btyukodi, 2017-01-09:
@neurodroid Thanks a lot, it works now! I also sent a message to the bug report.

neurodroid, 2017-01-10:
@btyukodi 
tl;dr: Even better would be to replace `--without-biosig` with `--with-biosiglite`

Long story: stimfit uses the biosig library for file i/o. The default version of biosig in the Debian repositories is heavily outdated and doesn't play along nicely with stimfit. You can use stimfit's own file i/o library instead by configuring `--without-biosig`, but this library has several issues of its own (see your other issue #55). To get a more recent version of biosig on Debian and address both of these problems, I'm now shipping stimfit with a minimal version of biosig (biosiglite) that you can enable by configuring `--with-biosiglite`. This flag is enabled by default in more recent deb package builds, but they haven't found their way into the repositories yet ([see this Debian bug report](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=847526)).

neurodroid, 2017-01-10:
@btyukodi [neurodebian ](http://neuro.debian.net/index.html) have backported stimfit 0.15.3 to xenial. If you [add their repository to your list](http://neuro.debian.net/index.html#get-neurodebian) you won't need to build from source.

btyukodi, 2017-01-10:
@neurodroid Great!!

closed 2017-01-10 by neurodroid