neurodroid/stimfit #55: CFS format sampling rate

btyukodi, 2017-01-10:
When I open the attached .cfs files in Signal 4.06 and check the View --> Channel information, it says that the sampling rate is 5kHz for one and 10 kHz for the other.
However, when I open them with the stfio module, the dt attribute is 0.0 for both cases. I'm using the latest version of stfio. What would be the proper way to extract the sampling frequency?

Thanks,
Botond

[cfs_samples.zip](https://github.com/neurodroid/stimfit/files/695412/cfs_samples.zip)

neurodroid, 2017-01-10:
Could you try to replace `--without-biosig` with `--with-biosiglite` while configuring?

btyukodi, 2017-01-10:
@neurodroid: Sure. Do I configure it once again and install it over the previous installation or do I have to remove it first? If the latter, how would I remove it?

btyukodi, 2017-01-10:
@neurodroid: Thanks a lot for your help. --with-biosiglite solve the problem, now I can access the sampling rate via the dt attribute.

closed 2017-01-10 by neurodroid