neurodroid/stimfit #59: Convert .abf to .ibw: Traces with variable length?

stbarlow24, 2017-07-12:
Hello,

I am an extreme beginner trying to use stimfit to convert a large number of gap-free, variable-length .abf files to .ibw files.  When I select the subdirectory of files to convert, I get an error message saying that 

> Files can't be exported. Traces have different sizes.

How can I get around this?

P.S. For reference, many traces have >500000 data points.  I assume the trace length problem is due to the last trace in each dataset being a bit shorter than the rest.

neurodroid, 2017-07-13:
Looks like we can only write Igor Binary Waves if all traces have the same size (not sure why - the code was written more than 10 years ago).
A tedious workaround would be to select all traces, then concatenate them, then export to Igor. You'd end up with a single massive wave.

schloegl, 2018-02-07:
@stbarlow24: did you find a workaround for converting .abf to .ibw ? 

IgorPro supports also HDF5
https://www.wavemetrics.com/products/igorpro/dataaccess/fileformats.htm

Is it an option to exporting your ABF data to hdf5 through stimfit, and importing it into IgorPro ?

schloegl, 2024-03-22:
There was no answer to the latest question. I assume the suggest workaround does solve this issue. If not, please reopen this issue.

closed 2024-03-22 by schloegl