neurodroid/stimfit #64: Window proportion and icon proportion issues in Stimfit (4k monitor)

Kemikal04, 2017-09-25:
I purchased a Lenovo T570 running Windows 10, with 3840x2160 (high res monitor).

Unfortunately, I am getting problems with window proportions in Stimfit. I have tried to start Stimfit with different compatibility modes, but that has only helped to very slightly improve the icon size.

See attachment:

[image]

Best regards,

Paul

neurodroid, 2017-09-25:
It seems like the library that we use for the GUI (wxWidgets) does not support high dpi displays.

It might be worth trying to build against wxwidgets 3.1, which should have better high dpi support (https://www.wxwidgets.org/news/2016/02/wxwidgets-3.1.0-released/). Unfortunately I don’t have access to a Windows machine at the moment.

If you have a Windows machine with Visual Studio 2008 64 bit installed I could guide you through the build process.

It has to be Visual Studio 2008 64 bit though. Any other version of Visual Studio won’t work.

See http://www.stimfit.org/doc/sphinx/win_install_guide/index.html

schloegl, 2025-07-12:
@Kemikal04 : Could you check whether this issue still persists in the latest version v0.16.8 of stimfit-lite ?