neurodroid/stimfit #68: Batch analysis with template matching

WouterDroogers, 2018-04-24:
Hi Stimfit team,

Is there a way to do template matching (e.g. the deconvolution method) on all selected traces at once?
In the GUI it is possible to do batch analysis with treshold crossings, but not template matching.
I can imagine it is possible to do this using a manually made function and/or using the Python shell. 
However, I'm not familiar enough with Stimfit and Python to figure out which existing functions I could use for this purpose. 
Could you point me in the right direction for a solution using either the GUI or the Python shell? 
Thanks in advance!

Best regards,

Wouter

WouterDroogers, 2018-04-24:
Thanks for your suggestion!
So there is not an easy way to put the deconvolution in a loop and extract events per trace?
Because, I can imagine that in your method cells or recordings/traces with a high number of events, have a bigger impact on the resulting parameters such as peak amplitude and rise-time. 
In my research I want to average the data of the cells in 1 group and not the data of all events of all cells in that group, which would be the result of concatenating all traces and then run the analysis. 
My apologies for bothering you with this and thanks for your time!

Best regards,
Wouter

neurodroid, 2018-04-24:
There’s an undocumented function criterion = stf.detect_events(template, mode=“deconvolution", norm=True, lowpass=0.5, highpass=0.0001)
that you could use from the Python shell. It computes the detection criterion for the currently active trace. You can set the currently active trace using stf.set_trace(i). In contrast to the GUI, you need to provide a template to this function. You could get this template for example from a previous fit using the second dimension of the result of stf.get_fit()
You could then detect events in your detection criterion using peak_idcs = stf.peak_detection(data, threshold, min_distance)
Where data would be the criterion in this case, threshold the appropriate threshold value to detect events, and min_distance the minimal distance between events (in number of sampling points).

WouterDroogers, 2018-05-07:
Thanks for your replies!
I have one question left:
How can I change the index of a trace? 
Because when I drag and drop my files into Stimfit, it automatically shifts the ordering of the recordings in such a way that the last selected recording has the lowest index in the Stimfit window (0). 
So I have the recordings A, B, C and D, which will have a shifted order in Stimfit: D, A, B and C. 

Best regards,
Wouter

neurodroid, 2018-05-14:
> How can I change the index of a trace?

You can set the index of a trace that you want to show using `stf.set_trace`

> Because when I drag and drop my files into Stimfit, it automatically shifts the ordering of the recordings in such a way that the last selected recording has the lowest index in the Stimfit window [...]

The order how files are opened during drag & drop are controlled by the operating system and the GUI toolkit that we use (wxWidgets). I don't think we have control over the order. If you wanted to open files in a reproducible order, I'd recommend to loop over a sorted file list and open them programatically with `stf.file_open`.

JoseGuzman, 2018-05-14:
Hi Wouter,

There are several posibilites to answer that question. I usually organise the recordings with the stimfit module (stfio) before using the GUI, to avoid using several recordings in one Stimfit program Take a look to the documentation [1] and let us know if this may work for you.

Alternatively, you can move one recording to the end of the list of opened recordings in Stifmit with the following trick:  Edit->Select all (in the recording you want to move) and Edit-> New Window with -> selected traces from this file. This will create a copy of the recording you selected to the end of the list. Unfortunately, it only works for recordings containing one channel.

You could also use stf.open_file() in stead of drag and drop files…

Let us know if you need further support.

Jose

[1] https://neurodroid.github.io/stimfit/stfio/index.html

JoSe GuZmAn
sjm.guzman@gmail.com

WouterDroogers, 2018-05-14:
Many thanks for your suggestions!
I might use `stf.file_open()` or `stfio.read()` in a loop to open all recordings in my sorted list. 
I will let you know when I need any further support.
For now I would like to thank you for your time!

Kind regards,
Wouter

closed 2018-05-28 by neurodroid