neurodroid/stimfit #70: stimfit not working on Ubuntu 18.04

raadnashmi, 2018-06-22:
Hi,
First thank you for making such a great software such as stimfit!
Stimfit was working fine on Ubuntu 16.04.  However, it is no longer working on Ubuntu 18.04.  The problem is that it does not start and I get the following error message:
"../src/common/object.cpp(251): assert "classTable->Get(m_className) == NULL" failed in Register(): Class "wxCommandEvent" already in RTTI table - have you used IMPLEMENT_DYNAMIC_CLASS() multiple times or linked some object file twice)?"
I get the identical error message on three other computers loaded with Ubuntu 18.04.  I would appreciate any help in this matter.
Thank you so much!
Raad Nashmi

neurodroid, 2018-06-28:
Problem ist that the default install of wxWidgets on bionic is `libwxgtk3.0` (gtk2) but `python-wxgtk3.0` appears to be linked against gtk3.
Solution is to install `libwxgtk3.0-gtk3-dev` and then configure as follows:
`configure --with-wx-config="wx-config --toolkit=gtk3"`
I've added `libwxgtk3.0-gtk3-dev` to `Build-Depends` in `debian/control` but I'm unsure how to enforce `configure --with-wx-config="wx-config --toolkit=gtk3"` in `debian/rules` if it is available. @yarikoptic any ideas?

yarikoptic, 2018-06-28:
smth like https://github.com/neurodroid/stimfit/pull/71 ? disclaimer: untested

raadnashmi, 2018-06-29:
Thank you, Christoph and Yaroslav.
Raad

neurodroid, 2018-06-29:
@yarikoptic Debian package fixing this issue is available on Debian mentors: https://mentors.debian.net/package/stimfit

neurodroid, 2018-06-29:
@raadnashmi The latest version was just uploaded from Debian mentors and should hit the Neurodebian servers soon. If you haven't done so already, the fastest way to get the fixed version is to add NeuroDebian to your list of repos: http://neuro.debian.net/index.html

closed 2018-06-29 by neurodroid

yarikoptic, 2018-06-29:
yep -- was uploaded to debian sid, and neurodebian repositories (main server is NH, the mirrors would get it within 24h), enjoy

raadnashmi, 2018-06-30:
> html

Christoph and Yaroslav,
Thanks so much!
Raad

Sverreg, 2019-12-12:
I see that this issue was resolved, but I still encounter the error on my ubuntu 18.04.

Installed using:
`sudo apt-get install stimfit python-stfio
`
Running stimfit gives the following error:
```
../src/common/object.cpp(251): assert "classTable->Get(m_classNsudo apt-get install stimfit python-stfio
ame) == NULL" failed in Register(): Class "wxCommandEvent" already in RTTI table - have you used IMPLEMENT_DYNAMIC_CLASS() multiple times or linked some object file twice)?
```

I installed `libwxgtk3.0-gtk3-dev `and configured it to `toolkit=gtk3`, but still encounter the same error when running stimfit.

[image]

My unix experience is very limited so I may have done something wrong, but I would have expected it to work out of the box.