neurodroid/stimfit #75: error when start Stimfit

athasd, 2019-01-08:
Hi,

we have just installed Stimfit (0.15.8) in a machine with windows 10, and when started we get the following error:

wxPython: stdout/stderr
C:\Python27\lib\site-packages\matplotlib\cbook\deprecation.py:107: MatplotlibDeprecationWarning: The mpl_toolkits.axes_grid module was deprecated in version 2.1. Use mpl_toolkits.axes_grid1 and mpl_toolkits.axisartist provies the same functionality instead.
  warnings.warn(message, mplDeprecation, stacklevel=1)

If we close the windows, we can work normally but then if we press create figure we also get:
one windows stating: couldn't create windows for the python shell
second windows stating: can not create figure (python/matplotlib is not available)

wxPython:stdout/stderr
Traceback (most recent call last):
  File "<string>", line 11, in plotWindowMpl
  File "embedded_mpl.py", line 41, in __init__
    self.canvas = FigCanvas(self, -1, self.fig)
  File "C:\Python27\lib\site-packages\matplotlib\backends\backend_wx.py", line 685, in __init__
    self.SetBackgroundColour(wx.WHITE)
  File "C:\Program Files\Stimfit 0.15\wx-3.0-msw\wx\_core.py", line 10863, in SetBackgroundColour
    return _core_.Window_SetBackgroundColour(*args, **kwargs)
TypeError: Expected a wxColour object, a string containing a colour name or '#RRGGBB', or a 3- or 4-tuple of integers.

any solution for this problem?
thanks

neurodroid, 2019-01-08:
A short-term dirty workaround would be to edit the file `C:\Python27\lib\site-packages\matplotlib\backends\backend_wx.py` and replace `wx.WHITE` with the tuple `(255, 255, 255)`. A long-term workaround is to update the Stimfit distribution for Windows. Hopefully I'll get to it at some point...

athasd, 2019-01-08:
Many Thanks!! 
it did not work with the tuple (255, 255, 255) but it worked with the string ('white')

closed 2019-01-08 by athasd