neurodroid/stimfit #77: unable to import stfio (MacPorts install)

snx27b, 2019-02-12:
Hi

I am trying to get stfio working on a new installation of MacPorts. I've installed python 3.5 and the py35-stfio port. When I try to use stfio within the python interpreter, I see the following. I am not sure if this is a bug or user error. If you have any suggestions or information I would hugely appreciate it.

Thank you,
Robert

```
$ python
Python 3.5.6 (default, Oct  3 2018, 19:57:36) 
[GCC 4.2.1 Compatible Apple LLVM 10.0.0 (clang-1000.11.45.2)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import stfio
Traceback (most recent call last):
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/stfio/stfio.py", line 20, in swig_import_helper
    return importlib.import_module(mname)
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 985, in _gcd_import
  File "<frozen importlib._bootstrap>", line 968, in _find_and_load
  File "<frozen importlib._bootstrap>", line 957, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 666, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 577, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 938, in create_module
  File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
ImportError: dlopen(/opt/local/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/stfio/_stfio.so, 2): Symbol not found: _min
  Referenced from: /opt/local/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/stfio/libbiosiglite.dylib
  Expected in: flat namespace
 in /opt/local/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/stfio/libbiosiglite.dylib

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/stfio/__init__.py", line 6, in <module>
    from .stfio import *
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/stfio/stfio.py", line 23, in <module>
    _stfio = swig_import_helper()
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/stfio/stfio.py", line 22, in swig_import_helper
    return importlib.import_module('_stfio')
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: No module named '_stfio'
>>> 

```

neurodroid, 2019-02-12:
Thanks for the bug report.

Fixed in 0.15.8 which I’ve just submitted to MacPorts:
https://trac.macports.org/ticket/58056
https://trac.macports.org/ticket/58057

Typically takes a couple of days until it hits all MacPorts repositories.

Christoph

snx27b, 2019-02-12:
Wow, thank you so much!!! I appreciate the fast response. 

Thanks for making this terrific software available! I used stfio for my PhD project and cited your 2014 publication in my paper (PMID 30228121).

Thanks very much,
Robert

neurodroid, 2019-02-18:
Thanks for using & citing Stimfit!
py-stfio has been updated in the MacPorts repositories. Can you update your local ports tree and let me know if it works now. There are still quite a bit of warning messages that we need to address, but the error that you report above should be fixed.

snx27b, 2019-02-18:
Hi Christoph,

Thanks so much for checking in! I noticed that it was updated last night and I was able to install it. I just ran my test data and it appeared to work well although I haven’t fully validated it yet. I am getting a warning about deprecated matplotlib functions but otherwise it seems fine. If I run into any additional issues I will let you know. Again, thank you so much for making this library available! It is incredibly helpful.

Thank you,
Robert

closed 2019-02-20 by neurodroid