neurodroid/stimfit #79: Add Python 3 support to stimfit

neurodroid, 2019-08-27:
stimfit uses the [wxWidgets](https://www.wxwidgets.org/) library for its user interface. The integration of the Python shell into stimfit's user interface strongly relies on the [`embedded`](https://github.com/wxWidgets/wxPython/tree/master/samples/embedded) sample that shipped with older versions of [wxPython](https://wxpython.org/), which is a Python interface to wxWidgets. These older versions of [wxPython](https://wxpython.org/) (codename 'classic') only support Python <= 2.7.
The [`embedded`](https://github.com/wxWidgets/wxPython/tree/master/samples/embedded) sample is no longer available in newer versions of wxPython (codename 'phoenix') that include support for Python 3.
I've discussed this topic back in 2013 (!) on the wxPython mailing list: https://discuss.wxpython.org/t/embedded-sample-for-phoenix/32518
Unfortunately I've never followed up on this issue.
To add Python 3 support to stimfit, we'd need an embedded sample for newer version of 
 wxPython ('Phoenix').

neurodroid, 2019-08-27:
> I have a few questions on this topic.
> 
> I'm assuming Stimfit still uses Classic since obviously it still has Python shell support. How much work would you think it would take to port over to Phoenix, assuming the embedded sample was implemented?

I expect the workload to be reasonable. Back in 2013 I tried to use Phoenix and got it to the point where everything compiled.

> For comparison, what would be the workload to change frameworks entirely? I remember this was [suggested (half-jokingly)](https://github.com/neurodroid/stimfit/issues/23) a few years ago. But I do see some appeal in the idea. In particular, I'm wondering whether it would be possible/practical to reimplement Stimfit in pure Python.
> 
> I see a few benefits to pure Python. First is the obvious opportunity to move to a more well-supported framework (I don't know if wxPython has improved over the last few years, so this may be a moot point).
> Second would be the expansion of the community of potential contributors. With the core GUI in C++, anyone who might contribute needs at least some familiarity with C++. But having the core GUI in Python would lower the barrier to contributing. In my case, I originally learned Python specifically to use the embedded shell in Stimfit. I'm now at a point where I think I am competent enough with Python to contribute to a Python project, but I am certainly not in a place to contribute to something in C++. I'm guessing I'm not alone here, so I think it's worth considering a port to pure Python to create a larger contributor community.

Re-writing Stimfit in pure Python would amount to writing a completely new program. I would not have the resources to support such a project.

There's also a non-negligible user base that uses the pure C++ version of Stimfit without any Python components. The advantage of this version is that it's extremely small and can be cross compiled for pretty much any host system in a matter of minutes with minimal dependencies.

neurodroid, 2019-08-27:
Your best bet to get Python 3 support for stimfit is to [voice your support for the corresponding issue on wxPython](https://github.com/wxWidgets/Phoenix/issues/1336).

They have 237 open issues so they naturally need to rank these by importance. The same thing happens with Stimfit: I can only address issues if I'm sure that a large enough user base will benefit from the work.

neurodroid, 2019-09-04:
[image]

neurodroid, 2019-09-05:
You can now give the Python 3 version a whirl on recent versions of Debian-based systems. Make sure that you build in a clean directory to avoid any issues with other versions.
`sudo apt-get install python3-wxgtk4.0`
`PYTHON=/usr/bin/python3 ../../configure --with-biosiglite --with-wx-config="wx-config --toolkit=gtk3"`

schloegl, 2019-09-06:
In addtition, I had to install 
    apt install python3-sip-dev
as well. Then,  ./configure ... && make  was working fine for me too.

analkumar2, 2019-10-04:
I am using windows subsystem for Linux Ubuntu. 
After doing 'git clone https://github.com/neurodroid/stimfit.git', I do 'cd stimfit'. Then when I run './configure' I get this error - '-bash: ./configure: No such file or directory'. Although when I do 'ls' it shows a file named 'configure.ac'.
Please help. I am trying to compile it for use in python3. autoconf is already installed.

schloegl, 2019-10-04:
configure can be generated with one of these commands: 
  ./autogen.sh 
  autoconf

neurodroid, 2019-10-04:
Interesting approach. I’d be curious if this works, though I thought that WSL does not support X? It might be difficult to get the graphical user interface to run.

At any rate, the configure script can be generated by using
./autogen.sh

analkumar2, 2019-10-04:
Thanks a lot. Here are the steps that worked for me:

sudo apt-get dist-upgrade
sudo apt-get install build-essential git libboost-dev python-dev python-numpy libhdf5-serial-dev swig libfftw3-dev libfftw3-doc fftw3 autoconf m4 python3-sip-dev python3-wxgtk4.0 libwxgtk3.0-gtk3-dev libhdf5-serial-dev
python3 -m pip install imutils pathlib2 wxPython --user
./autogen.sh
mkdir build
cd build
mkdir module
cd module
PYTHON=/usr/bin/python3 ../../configure --with-biosiglite --with-wx-config="wx-config --toolkit=gtk3" --enable-module --prefix=/usr
make
sudo make install
make clean

Some of the steps may be unnecessary and I don't know if '/usr' is correct for --prefix option. The stfio package now works for me in python3.
I won't be using the gui so not worried about stimfit. Although, it throws this when running stimfit from terminal:
[image]

With regard to X, Xming X server works for graphical interfaces with WSL.
Thanks a lot for developing stimfit.

neurodroid, 2019-10-04:
This problem typically means that wxPython and wxWidgets use different library versions.

See https://bugs.launchpad.net/ubuntu/+source/stimfit/+bug/1778433

Why do you install wxPython through both apt and pip?

I’d recommend to delete the pip version.

JoseGuzman, 2020-12-17:
Notes to myself:

* 1 Check python3 in system (without conda)
```bash
conda deactivate
which python3 # /usr/bin/python3
python3 --version # Python 3.6.9
sudo apt-get install python3-wxgtk4.0
```

* 2 After ./autogen, I run the configure script in a brand new stimfit location
```
PYTHON=/usr/bin/python3 ../../configure --with-biosig2 --with-wx-config="wx-config --toolkit=gtk3"`
```
and get this error:
```In file included from ./gui/app.cpp:51:0:
./gui/./app.h:209:10: fatal error: wxPython/wxpy_api.h: No such file or directory
 #include <wxPython/wxpy_api.h>
```

* 3 Test which wx-config we have:
```
wx-config --list
```
It returns:
```
Default config is gtk2-unicode-3.0
Default config will be used for output
Alternate matches:
base-unicode-3.0
gtk3-unicode-3.0
```

* 4  Let's see if I can set the wx_version
```
wx-config --prefix=/usr --version="gtk3-unicode-3.0"
```

It returns:
```
Warning: No config found to match: /usr/bin/wx-config --prefix=/usr --version=gtk3-unicode-3.0
in /usr/lib/x86_64-linux-gnu/wx/config
If you require this configuration, please install the desired
library build.  If this is part of an automated configuration
test and no other errors occur, you may safely ignore it.
You may use wx-config --list to see all configs available in
the default prefix.
```

* 5 Try to update the default configuration to gtk3

``` sudo update-alternatives --config wx-config
There are 3 choices for the alternative wx-config (providing /usr/bin/wx-config).

  Selection    Path                                                  Priority   Status
------------------------------------------------------------
* 0            /usr/lib/x86_64-linux-gnu/wx/config/gtk2-unicode-3.0   308       auto mode
  1            /usr/lib/x86_64-linux-gnu/wx/config/base-unicode-3.0   306       manual mode
  2            /usr/lib/x86_64-linux-gnu/wx/config/gtk2-unicode-3.0   308       manual mode
  3            /usr/lib/x86_64-linux-gnu/wx/config/gtk3-unicode-3.0   307       manual mode
```

Select *3*, and now gtk-unicode-3.0 is default:

after that, repeating from steps 2 I get the same error:

schloegl, 2020-12-17:
I do not see this error. The only difference is that I've only  gtk3-unicode-3.0 installed (no gtk2-unicode-3.0, or base-unicode-3.0). 

Maybe there is still something left from an earlier installation  ? Try "sudo make uninstall; sudo ldconfig; make distclean; ./configure ... && make && sudo make install "

Sverreg, 2021-08-18:
Jumping in with a possible unrelated issue. Is there an up to date guide for how to install Stimfit on Ubuntu (20.04)? I tried to follow the build guide in the docs and the comment here from analkumar2. 

Ended up at this error when I run sudo make install (not sure how much of the output is relevant): 

[image]

I initially tried running `sudo apt-get install stimfit python-stfio` as suggested on this github repo, but it could not find python-stfio and stimfit would not run. Probably doing something wrong.

[image]

schloegl, 2021-08-19:
These a two separate issues, (1) is compiling from source, (2) is installing through the package manager. 
In the first case you are missing some prerequisites, the error message indicates that liblapack-dev is not installed. You need at least 
    apt install liblapack-dev 
and perhaps some other prerequisites as well. (see "What we need before we start" in https://neurodroid.github.io/stimfit/linux_install_guide/requirement.html). 

The second issue is, that stfio might not be build for python2 anymore on your platform, and it has been changed to "python3-stfio". So I suggest doing
      apt install python3-stfio

Sverreg, 2021-08-19:
Ok, thanks! I might try again to compile from source from scratch, and make sure I get all the prereqs. 

For (2) installing through the package manager `sudo apt-get install python3-stfio` runs but i still get the same `Couldn't load wxPython core API` when I try to run stimfit. 

I tried installing wxPython:
```
pip install -U \
    -f https://extras.wxpython.org/wxPython4/extras/linux/gtk3/ubuntu-20.04 \
    wxPython
```
And the stimfit GUI now runs (but with new errors, see below):
[image]

And when I try to open .abf files:
[image]

schloegl, 2021-08-19:
The instructions to compile from source https://neurodroid.github.io/stimfit/linux_install_guide/requirement.html a fairly up-to-date. There should be no need to install wxPython from any other than the standard repository of your distribution (actually, this could cause incompatibilities). 

There are additional recommendations, when you compile stimfit souces from recent git. 
1) specify that you are using python3
2) if you use ABF2 data formats, I recommend installing biosig 2.3.1 or later, because that fixes a potential scaling issue.  

That means, I recommend to run configure with these arguments : 
 ` ./configure PYTHON_VERSION=3  --with-biosig `

closed 2023-08-30 by neurodroid