neurodroid/stimfit #83: support for opening different types of .dat files

hiris25, 2020-01-09:
I'm trying to open a .dat file generated by Robocyte2+ software, but the format is not supported. I've tried a conversion to ASCII but this also doesn't open. Any chances support for these files could be added?

schloegl, 2020-01-19:
In order to support that data format, we'd need : 
- the specification of the data format, and 
- some example data sets. 
Then we can check the feasibility of supporting this format. 

If this is not an option (e.g. because the format specification is not available), please check whether you can convert the data to any format that is supported by biosig [1,2]
[1] https://biosig.sourceforge.io
[2] http://pub.ist.ac.at/~schloegl/biosig/TESTED

schloegl, 2020-08-15:
Concering your request of support for opening .dat from Robocyte2+ software, I wondering if you can provide some example data sets. Without this, we'll not be able to follow up your request,

schloegl, 2024-03-22:
Without any example data and/or specification of the data format, it is impossible to address this.

closed 2024-03-22 by schloegl