neurodroid/stimfit #86: Error on file open

Xenodius, 2020-09-24:
I installed Stimfit with the compiled installer on two different systems-- one Windows 10, another Windows 7. When I try to load a HEKA .dat file on Win10 I get "Error opening file Unknown or unsupported file type" every time-- .amp/.mrk/.onl/.pgf/.pul/.sol are included in the directory. On Win7 this happens on about half the .dat files, but even those that open on the Win7 machine fail every time on the Win10 machine. Nonetheless Nest-O-Patch and HEKA Patchmaster reads them just fine. I am using HEKA 2x903. 

PS: I really just wanted the pystfio module, to read HEKA files and convert to ABF with pyABF. I copied the pystfio module from the source to my local venv library, but I cannot import it (e.g. import pystfio or stfio -> ModuleNotFoundError) I am not an expert with python but I've loaded (and created) modules this way before. Am I missing something?

schloegl, 2020-09-25:
Please check also https://github.com/neurodroid/stimfit/issues/32 . Currently, only HEKA files with valid bundle header (a.k.a. HEKA v2, a.k.a. HEKA DAT2) are supported. As a workaround can open the file again with Patchmaster and save it with the option "Make Bundle File" (see Patchmaster Reference Manual); this will store the file in the "bundled heka format" (a.k.a. HEKA DAT2)

I do not know whether the HEKA2x903 stores the data. In order to test this it would be good to have an example file.

Xenodius, 2020-09-25:
Thank you Schloegl; I did not have much hope as I already had bundle files enabled, but found a very similar workaround. I tried to 'Open - Modify' the .dat files in Patchmaster, 'Update' file, then close file, try again in Stimfit and it doesn't work... however, if I open modify/update then create a new file or open modify another .dat file instead of File->Close the file or Patchmaster, it works in Stimfit just fine. Very strange. 

Probably the issue is with Patchmaster and this can be closed. I can send you example files if you like but I'd have to send you a dropbox link privately or something, smallest is ~220MB.

schloegl, 2020-09-25:
I suspect the problem is related to some 32 bit limitation (Stimfit for windows, especially the python binding uses still some 32 bit libraries). In the process of loading the files, they can need a lot more RAM, than the file size (typically a factor of 5, but depending on the data layout, when you have different sampling rates, it can be even more.

I would expect the problem is gone, when using a pure 64 bit version of stimfit on a machine with sufficient RAM. Unfortunately for windows, we do not have this.  This can be tested, by using a*nix platform (Mac, or Linux), I'd suggest Linux, which shows the most stable behaviour. You can upload an example file to
    https://seafile.ist.ac.at/u/d/1e95fd96190646809fe9/
and I would have a look into this.

Xenodius, 2020-09-25:
Thank you Schloegl, I'm uploading them now. 

This is unrelated to the main issue and probably due to my ineptitude so I don't want to open another, but I've had a lot of trouble trying to compile the stfio module on windows. Spoiled by pip and know little about compiling-- perhaps you could direct me to some resources. Alternatively, is it possible to build on Linux from that more detailed stfio build guide, then copy to Windows? 

So far I tried using MinGW/msys, and I had to run `autoreconf -i` before I could call ./configure, even though there was already a configure and configure.ac file in the source directory I got a `./configure: No such file or directory` error each time until I ran `autoreconf -i`. When running` ./configure`, I then get the following error: 
```
checking python extra libraries... Traceback <most recent call last>:
  File "<string>", line 1, in <module>
TypeError: unsupported operand type<s> for +: 'NoneType' and 'str'
checking python extra linking flags... Traceback <most recent call last>:
  File "<string>", line 1, in <module>
TypeError: unsupported operand type<s> for +: 'NoneType' and 'str'

checking consistency of all components of python development environment... no
configure: error:
  Could not link test program to Python. Maybe the main Python library has been installed in some non-standard library path. If so, pass it to configure, via the LDFLAGS environment variable.
===============================================================
ERROR!
You probably have to install the development version of the Python package for your distribution. The exact name of this package varies among them.
===============================================================
```
I get the same error again when trying `./configure LDFLAGS="-L/python38/lib"` for python38 in non-OS drive, D:/python38/lib.

schloegl, 2020-09-25:
I can confirm that the "BROKEN" file is an Heka DAT1 file and the "FIXED" one is the HEKA DAT2 file. Only HEKA DAT2 files are supported in Stimfit and Biosig (which is used in to read most of the data formats). Doing this conversion is essential, and the recommended workaournd. I consider this issue solved. 

Concerning your other question, I'm also looking for a solution to make biosig-python binding work in windows.

closed 2020-09-25 by schloegl