neurodroid/stimfit #88: use the stimfit in the windows

yuan596, 2021-03-13:
hello!
I have some problems about the stimfit. How can I install the stimfit and use the python-stfio module in the windows? What version of python needs to be？
[image]

and how can I directly use the python-stfio module in the python？

schloegl, 2021-08-19:
The stimfit installer for windows comes with its own python version. In stimfit 0.15.8, it is some python 2.7.14, and it is installed by default in c:\python27

schloegl, 2025-04-02:
If you use stfio, only to load data into python, you can use the approach described here: 
https://github.com/neurodroid/stimfit/issues/108#issuecomment-2764233463

You can use 
`   pip install biosig ` 
and [demo2.py](https://sourceforge.net/p/biosig/code/ci/master/tree/biosig4c++/python/demo2.py?format=raw) from [Biosig](https://biosig.sourceforge.net/) in order to load data into python. 

Of course this is not a full replacement of stfio, because this approach does not provide any additional analysis functions as in stimfit.