neurodroid/stimfit #89: Missing labels on scale bar

Sverreg, 2021-08-22:
The labels on the scale bar are missing in stimfit 0.16.0, on Ubuntu 20.04 (unless there is a way to toggle labels that I can't find).

[image]

This might be related to the errors I get when I run stimfit, from[ Issue #79 ](https://github.com/neurodroid/stimfit/issues/79#issuecomment-901766763)

[image]

neurodroid, 2021-08-22:
Might have something to do with dark mode, fonts might be white in dark mode.

neurodroid, 2021-08-22:
@Sverreg Can you check the box for the second channel ("Show reference") in the top left box, and test whether the scale bar labels are printed in red for the second channel? That would tell us whether the problem lies with the dark mode.

Sverreg, 2021-08-23:
> Might have something to do with dark mode, fonts might be white in dark mode.

Ahh, you are right. Changing Ubuntu to Light or Standard mode immediately reveals the labels on the scale bar. 

> @Sverreg Can you check the box for the second channel ("Show reference") in the top left box, and test whether the scale bar labels are printed in red for the second channel? That would tell us whether the problem lies with the dark mode.

The red labels for the second channel are visible in Dark, Light and Standard mode Ubuntu. 

Thanks!

closed 2022-12-03 by schloegl