neurodroid/stimfit #93: HEKA channel interleave information missing

AMikroulis, 2022-08-31:
Hello.

The current version fails with long multi-channel HEKA recordings (truncates recording).
There is an old issue with hekalib.cpp.

In stimfit/src/libstfio/heka/hekalib.cpp lines 153 to 155 read:
```
...
    int TrFiller2; /* SET16 */
    };
C_ASSERT(sizeof(TraceRecord) == 296);
...
```

`TrFiller2` is a 32-bit unsigned int giving the channel interleave size in bytes.
The following 4 bytes (missing, but not necessary -would increase `sizeof(TraceRecord)` to `300`) are another unsigned giving the relative offset (in bytes) to the next chunk of the same channel.

Of course this applies only to (long) recordings with >1 channels.

schloegl, 2022-11-02:
In order to debug this issue, we'd need see an example file, or you run some tests for us. Can you provide some example file ? 
Alternatively, you can try converting your heka file with biosig-tools to a GDF file in the following way:  
    save2gdf  testheka.dat testheka.gdf

Does the conversion work ? And if yes, can you load testheka.gdf into Stimfit ?

schloegl, 2023-01-28:
Please check whether a recent version of stimfit (0.16.2) shows the same issue - I assume the problem is gone when using a more rencent version of stimfit.

closed 2023-01-28 by schloegl