neurodroid/stimfit #94: batch analysis, fitting with (monoexponential, offset fixed to baseline) doesn't work as expected

zcq0730, 2022-10-26:
I try to fit traces with (monoexponential, offset fixed to baseline) under batch analysis.
The traces are extracted IPSCs events.
However, the offset seems to be not fixed to the baseline of individual trace.
In the batch analysis, the baseline was measured as well, and I want this baseline value to be used as the offset for the monoexponential fitting function as a constraint.

Could you please let me know whether this is possible under the newest version of stimfit?

If not, could you let me know how to parse the value to the fitting function in stimfit python console? Thank you very much!

JoseGuzman, 2022-10-27:
Yes, that's possible. Did you select all traces before performing the batch analysis?

zcq0730, 2022-10-27:
> Yes, that's possible. Did you select all traces before performing the batch analysis?

Dear Dr. Guzman, 

thanks a lot for your quick reply. Yes, I selected all traces before performing the batch analysis.  Each trace is 50 ms in length. I want to set the mean value from 0-10 ms data as baseline.  And peak will be detected in the range of 12-25 ms. Fitting start from the detected timepoint of peak to the timepoint of 25 ms. Attached is an example traces I extracted from my recording.
I ran the batch analysis, but the offset value is not fixed as expected. (I used monoexponential, offset fixed to baseline)
[TEST.txt](https://github.com/neurodroid/stimfit/files/9877378/TEST.txt)
Please change the file extension from txt to abf.

Thanks in advance!

Best regards,
Chuanqiang

schloegl, 2022-11-02:
Dear Chuanqiang, 

I checked your data set, and could not reproduce this issue. Most likely some cursor settings were not chosen in proper way. Attached are the cursor configuration settings (test.csr) which you can load with 
   Stimfit Menu -> Edit -> Cursor settings -> Load 

When applying to all sweeps, and selecting 
   Stimfit Menu -> Analysis -> Batch Analysis -> Select your parameters including "Fit results"
you can get the attached Table of results (test.ods) 
    
I've been testing this with the lastest version from the git repository tag v0.16.1debian (commit 929d364c19195230e3b), but I do not see a reason why it should not work with some earlier versions. 

[test.zip](https://github.com/neurodroid/stimfit/files/9920111/test.zip)

Let us know if this does not work as expected, and specify the exact version of stimfit your are using.

zcq0730, 2022-11-03:
Dear Alois,
[image]
thank you very much for your reply. I expect the Base for each trace equals the Offset. Labled in yellow. I expect these two columns identical. But in your file, they are also not identical. Did I misunderstand something?

Best,
Chuanqiang

JoseGuzman, 2022-11-09:
Hi @zcq0730 !

The `baseline` and mono-exponential `Offset` are not exactly identical because the calculations are different and rely on distinct time points (i.e., cursor locations).

* The `baseline` computes the median (or mean) between the two baseline cursors.
* The `Offset` is a parameter obtained after fitting a mono-exponential function that 'decays or rises' to the baseline. When the exponential tends to infinity, the value of the function is the Offset. When fitting the function, we can start with any initial guess to obtain parameters such as the decay time, amplitude, or Offset. However, starting with the baseline to reach the function offset makes more sense. Thus, the function offset tends to the baseline value, but it's not fixed to it. We justified in Fig. 4 of the paper why using measurements (such as the baseline, peak, etc.) as initial fitting parameters performs faster and better fits that starting at random initial guesses.

Note that both baseline and offset values are statistically similar. Run a t-test - note the same variance - to test if your samples' mean between the baseline and offset is different (n=747). The probability of being different is P = 0.0002331832594162582, and the two-sided t-statistic is 3.698069089727761. 
You can download the samples and cursor configuration [here](https://github.com/neurodroid/stimfit/files/9970210/test.zip)

Baseline and Offset could have the same floating point precision if 

1. Traces contain no noise. 
2. If you locate the baseline cursors to the most extreme value of the function. In an ideal (noise-free) mono-exponential function, you will locate the baseline cursors at the plateau of the function.

You may have realised that the fitting functions return many other parameters (e.g., offset, different amplitudes,  etc. ). I like using those as sanity-checks. Fitting parameters must be around the physiological values you expect for your events. In most cases, you only need the time constants. Other parameters like peaks, baselines, etc., are measured better estimated with the standard measurement of Stimfit.

Let me know if you need further help!

Jose

zcq0730, 2022-11-09:
Dear Jose,
thank you very much for your detailed reply. But I am using the function 1: (monoexponential, offset fixed to baseline), not the function 0:  (monoexponential).  Would these two functions behave the same under batch mode? I might need your further help. When I fit a single trace with function 1, the offset value would indeed be fixed to the manually setted baseline value. (which I got from the standard stimfit measurment from baseline region.) (See the following reply to Christoph). 
Best,
Chuanqiang

zcq0730, 2022-11-09:
Dear Christoph,
thanks a lot for the question. Yes, it is as expected. But I need to manually change the offset value to the baseline value before I press OK button. See example fit for the first Trace.
Best,
Chuanqiang

[image]

JoseGuzman, 2022-11-11:
I may need to double-check this issue more carefully. I assumed the fitting menu updated its parameters from the measurements.

I'm not at the Society for Neuroscience (SfN) meeting in San Diego and will be back at my office next 21 November. Let me check about it that week. We could arrange a video conference to discuss the details of it if necessary.

zcq0730, 2022-11-12:
That's great! Thanks a lot! Would be happy to have a video chat to discuss
it. Looking forward, and enjoy your trip. Best, Chuanqiang

neurodroid, 2022-11-14:
It sounds like this is a general issue with the fitting procedure that is not restricted to the batch analysis.
It seems that the offset is kept fixed as it should be.
However, it is not initialised to the current baseline value.
I.e. to fix this issue we should set the initial parameter estimate for the offset to the current baseline value.

neurodroid, 2022-11-15:
The commit mentioned above should fix this issue. I've made some more changes to the code to get stimfit up to date so that we can build it on current platforms. Not sure when I'll get my hands on a Windows PC though.

zcq0730, 2022-11-15:
Dear Christoph,
cool! Thanks a lot! Looking forward to try it out. Another question
related, would it be possible to determine the fitting data range from Peak
to 20% decay for individual event? So that the data used for fitting
reflects more to the quick decay portion of the event. If it is too much
efforts please ignore this question.
Best, Chuanqiang

Nov. 2022, 16:22:

closed 2022-11-28 by zcq0730