neurodroid/stimfit #95: Crash after opening trace

GyulaMaloveczky, 2022-11-08:
[crash (15).zip](https://github.com/neurodroid/stimfit/files/9964197/crash.15.zip)
I am extracting EPSC-s using the event detection tool. After I extract them and start browsing it crashes on a particular trace. For example here in this file on the 15th.
I have opened the fine in clampfit, and that traced looked normal.
What could cause the issue?

JoseGuzman, 2022-11-11:
Would you please report us the Stimfit version and your operating system?

GyulaMaloveczky, 2022-11-12:
I am using windows 10 and the latest stimfit version

schloegl, 2022-12-03:
I checked your example file, and indeed there is a bug in the biosig import. This is fixed in 
    https://sourceforge.net/p/biosig/code/ci/94166059363d9f06670d178f745142785a16f85c/

At the moment I do not have a working toolchain to rebuild a full version of stimfit with embedded python for Windows. As a workaround, I can provided the following:
1) stimfit (w/o python) for windows 64bit
2) the converter "save2gdf.exe" for converting your data to a format that is readable by older versions of stimfit
3) The converted data file 
    save2gdf 'crash (15).atf' 'crash (15).gdf'

These files are available from here:
   https://seafile.ist.ac.at/f/9373808f79e341a0bd80/?dl=1

schloegl, 2023-01-28:
The problme should be fixed with stimfit 0.16.2 or later. Let us know if this is not the case.

closed 2023-01-28 by schloegl