neurodroid/stimfit #97: changing scale while reading abf files

GyulaMaloveczky, 2022-11-29:
I am opening abf files with stimfit (ABF 1.8 integer).
The Y scale is different. I was hoping that it would be only multiplied by a constant, and I could just calcualate the original values, however that is not the case. Usually the consant is around 400, but when I've made an abf file with integer values, like 1, or 2, the ratio of the stimfit and the clampfit (the standard abf viewer software) is completily different.
How could I transfer the actual Y values from an abf file to stimfit?
Something similar happens with the X axis, but since I know the sampling rate I can simply fix the issue.

schloegl, 2022-12-03:
Which version of stimfit are you using ? Is it using importBiosig, with with version of libbiosig ? 
You can check this after opening file and checking Stimfit Menu -> File -> File Information. 

I suspect it does not show "importBiosig with libbiosig 2.0 or later," , in that case you should upgrade to a newer version (I know that is a problem on windows). If you have a recent version of stimfit, please send the file, including a screenshot how the signal is displayed in clampfit.

GyulaMaloveczky, 2022-12-03:
Thank you! Yes it says biosig v1.9.
My stimfit is 0.15 by the way.
You mean I should update stimfit, right?

schloegl, 2022-12-03:
On which platform are you ? If you are on windows, I suggest to test this with the stimfit-lite (no python) . The most recent binary is available from this zip file:
   https://seafile.ist.ac.at/f/9373808f79e341a0bd80/?dl=1
It is not a full version of stimfit, because it does not include python. It is a portable stimfit.exe (no installation needed), und should be sufficient for testing the scaling. 

If the scaling is still wrong, I'd need your data file, and a screenshot from clamfit

schloegl, 2023-01-28:
Please check whether a recent version of stimfit (0.16.2) shows the same issue - I assume the problem is gone when using a more rencent version of stimit,

schloegl, 2024-03-22:
This has been fixed in biosig 2.5.1 and later, all Stimfit versions of 0.16.3 and later have this fixed.

closed 2024-03-22 by schloegl