neurodroid/stimfit #99: 'Error: Failed to build stimfit: command execution failed'  - MacOS Ventura 13.0

alinaaaltonen, 2023-02-26:
Could you verify whether I'm doing something wrong, or can it be there's an issue related to the OS?
Below is end part of the output when I run the install of stimfit: 

--->  Attempting to fetch stimfit-0.15.8.tar.gz from https://distfiles.macports.org/stimfit
--->  Verifying checksums for stimfit
--->  Extracting stimfit
--->  Configuring stimfit
--->  Building stimfit                              
Error: Failed to build stimfit: command execution failed 
Error: See /opt/local/var/macports/logs/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_science_stimfit/stimfit/main.log for details.
Error: Follow https://guide.macports.org/#project.tickets if you believe there
is a bug.
Error: Processing of port stimfit failed
--->  Some of the ports you installed have notes:
  OpenBLAS has the following notes:
    This version is built based on a base architecture for convenience,
    which may not be optimized for your system. To build a version
    customized for your machine, use the +native variant
  hdf5 has the following notes:
    Mac users may need to set the environment variable "HDF5_USE_FILE_LOCKING"
    to the five-character string "FALSE" when accessing network mounted files.
    This is an application run-time setting, not a configure or build setting.
    Otherwise errors such as "unable to open file" or "HDF5 error" may be
    encountered.
  lzma has the following notes:
    The LZMA SDK program is installed as "lzma_alone", to avoid conflict with
    LZMA Utils
  py27-cython has the following notes:
    To make the Python 2.7 version of Cython the one that is run when you
    execute the commands without a version suffix, e.g. 'cython', run:
    
    port select --set cython cython27
  py27-matplotlib has the following notes:
    The default backend is the interactive Mac OS X backend. Different backends
    can be specified using the ~/.matplotlib/matplotlibrc file. More details
    regarding backends can be found in the matplotlib FAQ:
    
            https://matplotlib.org/stable/users/explain/backends.html
  python27 has the following notes:
    To make this the default Python or Python 2 (i.e., the version run by the
    'python' or 'python2' commands), run one or both of:
    
        sudo port select --set python python27
        sudo port select --set python2 python27
  python310 has the following notes:
    To make this the default Python or Python 3 (i.e., the version run by the
    'python' or 'python3' commands), run one or both of:
    
        sudo port select --set python python310
        sudo port select --set python3 python310

alinaaaltonen, 2023-02-26:
the above is using the macports method. 

I tried also via homebrew to install the lite version and this also fails, below are the last lines: 

Installing schloegl/biosig/stimfit
==> Patching
==> ./autogen.sh && autoconf && automake
==> ./configure --prefix=/opt/homebrew/Cellar/stimfit/0.16.0 PYTHON_VERSION=3 --with-biosig --with-pslope
==> make PYTHON_VERSION=3
Last 15 lines from /Users/alina/Library/Logs/Homebrew/stimfit/03.make:
      __end_cap_(nullptr, __default_init_tag())
      ^
/Library/Developer/CommandLineTools/SDKs/MacOSX13.sdk/usr/include/c++/v1/vector:537:14: note: in instantiation of member function 'std::__vector_base<char, std::allocator<char> >::__vector_base' requested here
    explicit vector(size_type __n);
             ^
./intan/streams.cpp:186:22: note: in instantiation of member function 'std::vector<char>::vector' requested here
        vector<char> tmp(size + 2);
                     ^
2 errors generated.
make[3]: *** [streams.lo] Error 1
make[3]: *** Waiting for unfinished jobs....
libtool: compile:  clang++ -DHAVE_CONFIG_H -I. -I../.. -DWITH_PYTHON -DWITH_PSLOPE -DWITH_BIOSIG -DWITH_BIOSIG -DHAVE_LAPACK -DHAVE_LAPACK_SUFFIX -DH5_USE_16_API -g -O2 -fPIC -Wall -c ./intan/intanlib.cpp -o intanlib.o >/dev/null 2>&1
make[2]: *** [all-recursive] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/homebrew-core):
  https://github.com/schloegl/homebrew-biosig/issues

neurodroid, 2023-03-24:
@alinaaaltonen For the MacPorts version, I've submitted a ticket to MacPorts that should fix this issue:
https://trac.macports.org/ticket/66271
However, this ticket has been pending for 4 months now.
There are 2 options for you to get stimfit running with MacPorts:
Either you use the patch that comes with the ticket mentioned above.
Alternatively you try to get the MacPorts maintainers to apply the patch by posting to the ticket mentioned above.

alinaaaltonen, 2023-03-26:
Thanks for your replies.  I'm afraid I'm not very experienced with programming in general but I tried to apply the above fixes.  Apologies if I make mistakes or if I can't convey my issues well.

@neurodroid regarding the macports version, I noticed that trying to build py37-stfio with macports gives me a similar error message, and when I run ./configure in the stimfit directory I get the error message 

...
checking for python... no
configure: error: Cannot find python in your system path

I am not sure how to apply the patch you mentioned, as I checked the stimfit portfile and it seems to be the same as the one in the ticket.  
But I am wondering if it may be related to python?

alinaaaltonen, 2023-03-26:
@schloegl again apologies if it's maybe my lack of knowledge, but I wasn't sure exactly how to apply the patch you mentioned.  I tried a few times to run ./configure both with and without CC=gcc-11 CXX=g++-11 (gcc@10 didn't work) and instead here I get the error message 

configure: error: cannot find install-sh, install.sh, or shtool in "." "./.." "./../.."

and I am not sure how to continue from here.

ryandesign, 2023-03-28:
> I've submitted a ticket to MacPorts that should fix this issue:
> https://trac.macports.org/ticket/66271
> However, this ticket has been pending for 4 months now.

Apologies that we missed that ticket for so long. I've dealt with it now.

alinaaaltonen, 2023-03-28:
@ryandesign @neurodroid I wiped the old files and tried again to build via Macports but unfortunately it still did not work. I get the same error message as above.

neurodroid, 2023-03-28:
@alinaaaltonen Make sure you have the latest portfile. Did you run `sudo port selfupdate` ?

alinaaaltonen, 2023-03-28:
@neurodroid Sorry I should have mentioned that before, yes I have an M2-chip MacBook Pro.  I didn't realise that would make a difference.

ryandesign, 2023-03-30:
The fix for #100 is now in MacPorts, if you'd like to selfupdate and try again.

alinaaaltonen, 2023-03-31:
@ryandesign thank you so much, the build finally worked! 

There do seem to be some bugs, from some very initial playing around I found that the files are being automatically concatenated (nice for free run, but not so good for other recording types). Also when I try to open the window for cursor settings I get an error message 'a debugging check in this application has failed'. Finally I noticed that the 'f' fit trace button does not fix properly to the screen.  

Do you have any suggestions for what I might need to fix @neurodroid ?

neurodroid, 2023-03-31:
Glad you got it to work. I'll cose this issue for now.

As far as the bugs are concerned, please open separate issues for each of them, with step-by-step instructions for reproducing them, including the data file that you used when you encountered the issue.

closed 2023-03-31 by neurodroid