neurodroid/stimfit #102: ABF file viewing issues in Stimfit 16.2 on Mac

alinaaaltonen, 2023-04-04:
I recently managed to build stimfit on my Mac with M2 chip on Ventura OS, via Macports.  I have now several issues (see other issues for more details).  Firstly, I have found that when I open ABF files with multiple sweeps, the sweeps are automatically concatenated into one trace.  In addition, when using the 'fit traces to window' function, the borders of the trace exceed the field of view.

neurodroid, 2023-04-05:
Can you please provide us with an abf file so that we can reproduce this issue

alinaaaltonen, 2023-04-05:
Of course, here is an example: 
[210423001_baseline.abf.zip](https://github.com/neurodroid/stimfit/files/11163009/210423001_baseline.abf.zip)

neurodroid, 2023-04-06:
@schloegl Here is the output from biosig (using `configure --with-biosig`):
```
Warning ABF2 v2.90: nOperationMode=5 is very experimental - double check the data you get.
Number of events: 0

===========================================
[FIXED HEADER]
Recording:
	ID              : 00000000
	Institution     : (null)
	Technician      : cs	# default: localuser
	Equipment       : b4c_3.0 
	IPv4 address    : 0.0.0.0	 # default:local host
Manufacturer:
	Name            : (null)
	Model           : (null)
	Version         : (null)
	SerialNumber    : (null)
Patient:
	ID              : 
	Gender          : unknown
	Eye Impairment  : unknown
	Heart Impairment: unknown
	Age             : ----
	Birthday        : unknown
	StartOfRecording: (738269.610093) 2021-04-23 14:38:31.999986	Timezone        : +120 min

bpb=4
row-based=0
uncalib  =0
OFdetect =1
FileName:	/Users/cs/Downloads/210423001_baseline.abf
Type    :	ABF2
Version :	2.90
HeadLen :	18432
NoChannels:	1
SPR:		1
NRec:		1365000
Fs:		10000.000000
Events/Annotations:	0
Events/SampleRate:	10000.000000

[CHANNEL HEADER] 0x7fa3ff28abf0
No  LeadId Label	Fs[Hz]	SPR	GDFTYP	Cal	Off	PhysDim	PhysMax 	PhysMin 	DigMax  	DigMin  	HighPass	LowPass 	Notch   	delay [s]	X	Y	Z
#01:   0 0 unspecified      	10000.000000     1 float32  1.907356e-03 0.000000e+00 ?	16000.1	-16000.1	8388607.000000	-8388607.000000	1.000000	5000.000000	0.000000	    0	0.000000	0.000000	0.000000

[EVENT TABLE 0] N=0 Fs=10000.000000
No	TYP	POS	CHN	DUR/VAL	Desc

```

neurodroid, 2023-04-06:
@schloegl biosig indeed loads this file as a single sweep.
Using `configure --without-biosig --without-biosiglite` (i.e. using the legacy abf code from libstfio), the file is loaded with 91 sweeps

schloegl, 2023-04-26:
A fix for this issue is in biosig is provided by this this commit. 
https://sourceforge.net/p/biosig/code/ci/4bbbcf3b906092ee316f6331f19c96757b5c6a12/

alinaaaltonen, 2023-04-26:
I updated and reinstalled stimfit and its dependencies and the issue persists for me. Is there something I might have done wrong?

neurodroid, 2023-04-26:
All these changes were only applied in the code repository. It will take a while until they make it to MacPorts, as we'll have to release a new version first.

schloegl, 2023-07-23:
ABF2 data is now loaded either with libbiosig 3.0.2 (as released in biosig 2.5.1 or later), or with internal ABF2 reading functions. According to my tests, this fixed this issue.

closed 2023-07-23 by schloegl