neurodroid/stimfit #103: wxWidgets Bebug Alert - Stimfit 16.2 on Mac

alinaaaltonen, 2023-04-04:
I recently managed to build stimfit on my Mac with M2 chip on Ventura OS, via Macports.  I have now several issues (see other issues for more details).  The second, more major issue that I have found is that when I try to open the cursor settings window or  other analysis windows I get this error message 'a debugging check in this application has failed' and the application then crashes.  I did my best to capture the message for more details. 

[image]

neurodroid, 2023-04-05:
I can reproduce this issue. It seems to be caused by wxWidgets now reporting debug messages even if it's not explicitly built as a debug version. It's not a fix to this problem, but turns out that repeatedly chosing "continue" instead of "stop" in the debug dialog  lets you get to the actual dialog in the end.

alinaaaltonen, 2023-04-05:
I updated the port tree and tried again, I still get the error message but I see if I click 'don't show this dialog again' and then continue it works.

neurodroid, 2023-04-06:
The code changes will only make it into MacPorts once a new version of stimfit is released.

closed 2023-08-17 by schloegl