neurodroid/stimfit #104: Y axis scale issues in ABF files Stimfit 16.2 Mac

alinaaaltonen, 2023-04-06:
Another issue that I have now identified is that the Y-axis scale is incorrect, such that peaks that should be 100mV show as something like 0.0001, and the scale bars are not working as intended, which is most likely related. Here is a file for testing: 
[221021005_rheo.abf.zip](https://github.com/neurodroid/stimfit/files/11167034/221021005_rheo.abf.zip)

neurodroid, 2023-04-06:
@schloegl This is an issue with biosig. Using the legacy abf code from libstfio, the scaling is read out correctly:
[image]
Whereas using biosig (`configure --with-biosig`), the scale is incorrect:
[image]

schloegl, 2023-04-29:
The patch allows to compile --with-biosig and get all the advantages of biosig, but switches reading of ABF2 from biosig to the internal abf reading functions. 
When some future version of libbiosig has its ABF2 support fixed, this can be reverted.

schloegl, 2023-07-23:
ABF2 data is now loaded either with libbiosig 3.0.2 (as released in biosig 2.5.1 or later), or with internal ABF2 reading functions. According to my tests, this fixed this issue.

closed 2023-07-23 by schloegl