neurodroid/stimfit #108: cfs files recorded with Signal version 4.11 loaded incorrectly

merdor, 2025-03-18:
Hello,
I am using Signal version 4.11 to record patched pyramidal neurons. The cfs files recorded on one of our setups are, unfortunately, imported incorrectly with stimfit.  

my code is the following:

````python
import neo
import numpy as np
import stfio
import matplotlib.pyplot as plt
import matplotlib
file = '20250314_S5_C1_CellChar.cfs'
stfio_obj = stfio.read(file, 'cfs')
neo_obj = neo.io.StimfitIO(stfio_obj)
spike_data = neo_obj.read()[0]
fig, ax = plt.subplots(nrows=2)
len_ana = np.min([len(spike_data.segments),20])
for cS in range(0,len_ana):
     cur = np.asarray(spike_data.segments[cS].analogsignals[0])
     vol = np.asarray(spike_data.segments[cS].analogsignals[2])
     ax[0].plot(cur)
     ax[1].plot(vol)
````

I have attached a screenshot of what I am getting in Signal and what I am getting in Python below. I have also attached the problematic CFS file (see [20250314_S5_C1_CellChar.zip](https://github.com/user-attachments/files/19322137/20250314_S5_C1_CellChar.zip)).

Interestingly, files from another setup seem to be imported correctly.

Best, 
Martin

[image]
[image]

CC: @ximion

schloegl, 2025-03-29:
The problem is fixed in biosig 3.9.0. Just make sure you have libbiosig v3.9.0 installed, and stfio is linked against it. 
 
I've been able to produce from your data these figures 
[image]
by using [https://pypi.org/project/Biosig/](https://pypi.org/project/Biosig/) and [demo2.py](https://sourceforge.net/p/biosig/code/ci/master/tree/biosig4c++/python/demo2.py?format=raw
) from [Biosig](https://biosig.sourceforge.net/).

closed 2025-03-29 by schloegl

ximion, 2025-03-29:
Awesome, thank you so much for the quick fix! Do you plan to update the Debian package @ Debian too?

schloegl, 2025-03-30:
Sure, but because the release process in Debian is more elaborate, I can not say when it will be available, and whether it will make it into Debian13/trixie (see [0] for details). 

However, you can install it also from source following this reprocedure [1]. You just need to download [2] and extract the file 

tar xvf biosig-3.9.0.src.tar.xz
cd biosig-3.9.0
./configure && make -j4 && sudo make install 

This will install libbiosig in /usr/local/lib which is in the LD_LIBRARY_PATH

Because Debian is using dynamic/shared libraries and the ABI/API of libbiosig has not changed, there is a good chance that you can even use Debian's stfio. 

[0] https://release.debian.org/trixie/freeze_policy.html  
[1] https://sourceforge.net/p/biosig/wiki/Installation/
[2] https://sourceforge.net/projects/biosig/files/BioSig%20for%20C_C%2B%2B/src/biosig-3.9.0.src.tar.xz/download

ximion, 2025-03-30:
Nice! If there's anything on the Debian side that I could help with (even if you need sponsorship), let me know :-) (I'm a Debian Developer). Getting this in before the freeze is definitely possible at this point, if done before 10. April.

schloegl, 2025-03-31:
Support for debian packaging is welcome, my proviciency in debian packaging has room for improvement. So I can not say whether I'll be able to package it for April 10th. 
If you like to step it, feel free to do so. You might want to coordinate also with the other Debian maintainer and uploaders.