neurodroid/stimfit #111: "Trace selection index"  not useful as input on stimfit-lite for windows

schloegl, 2025-06-26:
The field "Index" in the "Trace selection" panel can be also used for entering the trace number. 
Entering e.g. the number "5" should than show the 5 trace (a.k.a. sweep).

This is working well on the linux version of stimfit, however on the windows version of the stimfit-lite, you can enter that number, but chances are that the trace index is changing to some other number, and some other sweep is shown. Therefore, this field can not be used enter the desired sweep number. 

It would be desirable that the "Trace selection" field can be also used on windows for selecting the sweep number.

JoseGuzman, 2025-07-04:
* We need to decide whether to use a zero-based index or not. In this case, providing more possibilities (zero, and one index-based) reduces usability.  @neurodroid, do you have an opinion on this?

* The Linux vs Windows performance repeats frequently, because everyone is developing on their local machines. I would suggest creating a Docker version for Windows/Linux, so that we can test/debug/develop on our local machines.

closed 2025-07-09 by schloegl

JoseGuzman, 2025-07-09:
Thanks for closing this. Just noting: using containers would also benefit Mac, Linux, and cloud users by ensuring consistent environments, easier dependency management, and alignment with CI/CD pipelines. Worth considering for future development beyond Windows.